mne-tools/mne-python #12286: Consider "Publishing to PyPI with a Trusted Publisher"

hoechenberger, 2023-12-10:
> This method can be used in automated environments and eliminates the need to use username/password combinations or manually generated API tokens to authenticate with PyPI when publishing.
> ...

>1. Certain CI services (like GitHub Actions) are OIDC identity providers, meaning that they can issue short-lived credentials ("OIDC tokens") that a third party can strongly verify came from the CI service (as well as which user, repository, etc. actually executed);
> 1. Projects on PyPI can be configured to trust a particular configuration on a particular CI service, making that configuration an OIDC publisher for that project;
> 1. Release automation (like GitHub Actions) can submit an OIDC token to PyPI. The token will be matched against configurations trusted by different projects; if any projects trust the token's configuration, then PyPI will mint a short-lived API token for those projects and return it;
> 1. The short-lived API token behaves exactly like a normal project-scoped API token, except that it's only valid for 15 minutes from time of creation (enough time for the CI to use it to upload packages).
> 

https://docs.pypi.org/trusted-publishers/

larsoner, 2023-12-11:
I set this up for some repo recently (can't remember which!) following their instructions and it worked, so +1 for transitioning at some point 👍

drammock, 2023-12-11:
We already have this set up for our `release.yml` action.

[image]

closed 2023-12-11 by drammock

hoechenberger, 2023-12-11:
Oh, I missed that. Nice work!