mne-tools/mne-python #12290: Reading and writing TFRs is confusing

cbrnr, 2023-12-12:
I thought about asking in our forum, but I decided to open an issue, because I think this is at least partly a documentation issue:

I am confused by our TFRs API. In particular, it is not clear to me how reading and writing is supposed to work. There are two functions, [`mne.time_frequency.read_tfrs()`](https://mne.tools/stable/generated/mne.time_frequency.read_tfrs.html) and [`mne.time_frequency.write_tfrs()`](https://mne.tools/stable/generated/mne.time_frequency.write_tfrs.html).

1. Let's start with `write_tfrs()`. According to its docstring, I can pass `AverageTFR | list of AverageTFR | EpochsTFR` – a list of `EpochsTFR` is not supported, but I can still pass it without any problems. So I guess the line in the docstring should be adapted to `AverageTFR | list of AverageTFR | EpochsTFR | list of EpochsTFR`.

    BTW, and this should probably be a separate discussion, but I really think we should use type hint syntax in our docstrings. For example, the previous docstring should really be changed to `AverageTFR | list[AverageTFR] | EpochsTFR | list[EpochsTFR]`.
2. I have no idea what this note in the description of the `tfr` parameter means: "Note. If .comment is not None, a name will be generated on the fly, based on the order in which the TFR objects are passed."
3. Similarly, the docstring of `read_tfrs()` says that the function returns `AverageTFR | list of AverageTFR | EpochsTFR`, so this is probably also missing a `list of EpochsTFR`.
4. The docstring also says that the return value of `read_tfrs()` depends on `condition`: "Depending on `condition` either the TFR object or a list of multiple TFR objects." This is only partly true, because I also get a list of `AverageTFR` if I read a file containing only a single `AverageTFR` object (same for a single `EpochsTFR` object). So this comment needs to be adapted, or maybe even better, the function should *always* return a list (of `EpochsTFR` or `AverageTFR`), even when there is only a single TFR.
5. I don't understand the `condition` parameter. How are conditions stored in `AverageTFR` objects?
6. It seems like `condition` does not work for `EpochsTFR`, this should be documented. And because I don't understand it in the first place, why does it not work for `EpochsTFR`?
7. The types of `condition` should be changed to `int | str | list of int | list of str | None`.

Sorry for this long list!

larsoner, 2023-12-12:
>  have no idea what this note in the description of the tfr parameter means: "Note. If .comment is not None, a name will be generated on the fly, based on the order in which the TFR objects are passed."... The docstring also says that the return value of read_tfrs() depends on condition

I think the API is designed to mirror `read_evokeds`, with all its shortcomings. This includes the idea of there being a `.comment`, though that probably only applies to EpochsTFR. I think being able to save `list of EpochTFR` is an (unhappy) accident, but I guess we can live with it. +1 for updating the docstring to make it clearer what it actually does.

> or maybe even better, the function should always return a list (of EpochsTFR or AverageTFR), even when there is only a single TFR.

-1 on this just like for `read_evokeds`, where IIRC we had a similar discussion recently

cbrnr, 2023-12-13:
> I think the API is designed to mirror `read_evokeds`, with all its shortcomings. This includes the idea of there being a `.comment`, though that probably only applies to `EpochsTFR`.

So is `.comment` connected to `condition`? How do I store multiple conditions in a file (there is no option in `write_tfrs()`)? In addition, `condition` only works for `AverageTFR`, not for `EpochsTFR`.

> -1 on this just like for `read_evokeds`, where IIRC we had a similar discussion recently

OK, I have yet to see an example of it returning a single TFR object. So far, I have only ever received a list containing a single TFR object.

larsoner, 2023-12-13:
> So is .comment connected to condition? How do I store multiple conditions in a file (there is no option in write_tfrs())? In addition, condition only works for AverageTFR, not for EpochsTFR.

I cannot remember the details, someone will have to investigate

>  I have yet to see an example of it returning a single TFR object. So far, I have only ever received a list containing a single TFR object.

Ahh in that case if it already always returns a list then we should document the behavior properly