mne-tools/mne-python #12305: tfr_array_stockwell `data` argument does not match the `epoch_data` argument of tfr_array_multitaper and _morlet functions

tsbinns, 2023-12-18:
### Proposed documentation enhancement

Hi, just a small suggestion:

I noticed that that data is given to the [`tfr_array_multitaper`](https://mne.tools/stable/generated/mne.time_frequency.tfr_array_multitaper.html) and [`tfr_array_morlet`](https://mne.tools/stable/generated/mne.time_frequency.tfr_array_morlet.html) functions as `epoch_data`, but as `data` for [`tfr_array_stockwell`](https://mne.tools/stable/generated/mne.time_frequency.tfr_array_stockwell.html).

All three of these functions state that the data must be of shape `(n_epochs, n_channels, n_times)`, so it is clear that even for `tfr_array_stockwell`, `data` should be epoched, however this is not so clear from the Looking at the [Time-Frequency module API](https://mne.tools/stable/api/time_frequency.html).

I could see someone skimming the API thinking that `tfr_array_multitaper` and `tfr_array_morlet` must be performed on epoched data, but `tfr_array_stockwell` on raw or epoched data given the different name of the data argument.

Changing

larsoner, 2023-12-18:
+1 for changing the name. It's very likely people mostly have exclusively used it positionally so doing
```
def tfr_array_stockwell(epoch_data, ..., *, data=None, verbose=None)
```
and emitting a `FutureWarning` when `data is not None` should be safe enough. @tsbinns up for opening a PR to do this?

drammock, 2023-12-18:
just to play devil's advocate for a moment, would it be better to change the other two functions' first param to `data` instead?  To me it feels slightly better (because `data` as first param is a very common pattern, and not just for us). But I'm content to defer to @larsoner or others on this point

larsoner, 2023-12-18:
Works for me

tsbinns, 2023-12-18:
Okay, so switch the multitaper and Morlet functions to `data` and set `epoch_data=None` as default (raising a `FutureWarning`  if `epoch_data != None`

drammock, 2023-12-18:
> Okay, so switch the multitaper and Morlet functions to `data` and set `epoch_data=None` as default (raising a `FutureWarning` if `epoch_data is not None`)?

yep! thanks.

closed 2023-12-19 by larsoner