mne-tools/mne-python #12310: apply_hilbert for source estimated data

BabaSanfour, 2023-12-19:
### Describe the new feature or enhancement

Currently apply_hilbert can be applied only to raw/epochs/epoched data through, for example, epochs. apply_hilbert.
The idea is to change this whole function into the time_frequency module. Then the epochs/raw/evoked/source data can be an argument to the function  (EX: mne.time_frequency.EpochsTFR) 

### Describe your proposed implementation

It can be a new class in the tfr.py file. It would be based on the ancient .apply_hilbert function.

### Describe possible alternatives

The alternative that is being used now is script hilbert function for source data or apply_hilbert for other types.
The proposed implementation will make the function independent of the data type/object

### Additional context

I am happy to propose an implementation. I just want to check with maintainers for directions before starting. As this is a huge change in a lot of classes/files. I think it should be discussed before starting the implementation.

larsoner, 2023-12-19:
Alternatively we could make `FilterMixin` work for SourceEstimate instances. I think in principle it should be doable -- would need to make `FilterMixin.filter`, `FilterMixin.savgol_filter`, and `FilterMixin.resample` (which already exists for SourceEstimate classes) work properly. WDYT?

BabaSanfour, 2023-12-19:
Yes that would work too;
By "already exists for SourceEstimate classes" you refer to the resample function only right? could not find the other implementations.

larsoner, 2023-12-20:
Yeah just resample exists currently

BabaSanfour, 2023-12-23:
ok thanks! I will take that into account when making the changes.

BabaSanfour, 2023-12-31:
@larsoner the function has been added

PR #12323 (2024-02-13): Apply hilbert stc
Fixes #12310.

Updating mne.filter.MixinFilter methods to work with Source Estimated data
closed 2024-02-13 by larsoner