mne-tools/mne-python #12331: pip-pre CIs with numpy and matplotlib pre-release

mscheltienne, 2024-01-02:
### Description of the problem

When installing pre-releases from https://pypi.anaconda.org/scientific-python-nightly-wheels/simple:

> (1):  python -m pip install --upgrade --pre --only-binary :all: -i https://pypi.anaconda.org/scientific-python-nightly-wheels/simple --timeout=180 numpy scipy matplotlib
> (2): python -m pip install --upgrade --only-binary ":all:" --default-timeout=60 --extra-index-url "https://pypi.anaconda.org/scientific-python-nightly-wheels/simple" "numpy>=2.0.0.dev0" "scipy>=1.12.0.dev0" matplotlib

(1) will not install `numpy` 2.0; while (2) will. But, the import of `matplotlib` yields:

> A module that was compiled using NumPy 1.x cannot be run in
> NumPy 2.0.0.dev0 as it may crash. To support both 1.x and 2.x
> versions of NumPy, modules must be compiled against NumPy 2.0.
> 
> If you are a user of the module, the easiest solution will be to
> either downgrade NumPy or update the failing module (if available).
> 
> NOTE: When testing against pre-release versions of NumPy 2.0
> or building nightly wheels for it, it is necessary to ensure
> the NumPy pre-release is used at build time.
> The main way to ensure this is using no build isolation
> and installing dependencies manually with NumPy.
> For cibuildwheel for example, this may be achieved by using
> the flag to pip:
>     CIBW_BUILD_FRONTEND: pip; args: --no-build-isolation
> installing NumPy with:
>     pip install --pre --extra-index-url https://pypi.anaconda.org/scientific-python-nightly-wheels/simple
> in the `CIBW_BEFORE_BUILD` step.  Please compare with the
> solutions e.g. in astropy or matplotlib for how to make this
> conditional for nightly wheel builds using expressions.
> If you do not worry about using pre-releases of all
> dependencies, you can also use `--pre --extra-index-url` in the
> build frontend (instead of build isolation).
> This will become unnecessary as soon as NumPy 2.0 is released.
> 
> If your dependencies have the issue, check whether they
> have nightly wheels build against NumPy 2.0.
> 
> pybind11 note: You may see this message if using pybind11,
> this is not problematic at pre-release time
> it indicates the need for a new pybind11 release.

Observable in CIs here: https://github.com/mne-tools/mne-python/actions/runs/7362617021/job/20069242175?pr=12326#step:9:65

----------

The issue is that `matplotlib` requires `contourpy` which requires `numpy < 2.0`. I'm the surprised the line (2) is able to force the install. I don't know if this will impact MNE, and I don't know how to resolve this without relaxing the requirement of `contourpy` or without running `pip-pre` test against `numpy / scipy` first, and then against `matplotlib`. 

x-ref: https://github.com/contourpy/contourpy/issues/344

### Steps to reproduce

```Python
-
```

### Expected results

-

### Actual results

-

### Additional information

-

larsoner, 2024-01-02:
@mscheltienne can you open this with the matplotlib folks? Seems like they should update their prerelease wheel infrastructure to maybe also build contourpy and upload it to https://anaconda.org/scientific-python-nightly-wheels, as they seemingly must already build `contourpy` using NumPy 2.0 to upload wheels at all...?

larsoner, 2024-01-02:
Actually looks like https://github.com/matplotlib/matplotlib/issues/26847 was opened at one point for this

mscheltienne, 2024-01-03:
... but that does not work either since the job with `numpy` and `scipy` pre-release will also install `matplotlib` stable, compiled with `numpy 1.xx` and thus yielding the same warning.

mscheltienne, 2024-01-24:
Ended up with:

```
python -m pip install --progress-bar off --upgrade git+https://github.com/mne-tools/mne-python
python -m pip install matplotlib
python -m pip install --progress-bar off --upgrade --no-deps --pre --only-binary :all: -i https://pypi.anaconda.org/scientific-python-nightly-wheels/simple --timeout=180 matplotlib
python -m pip install --progress-bar off --upgrade --pre --only-binary :all: -i https://pypi.anaconda.org/scientific-python-nightly-wheels/simple --timeout=180 numpy scipy
```

Can't remember all of the rationale behind this.. but in the end this issue is now outdated.

closed 2024-01-24 by mscheltienne