mne-tools/mne-python #12333: [BUG] Interpolate bads ignores ECoG and sEEG channels

alexrockhill, 2024-01-03:
### Description of the problem

Interpolate bads silently ignores `ecog` and `seeg` channels.

### Steps to reproduce

```Python
import mne

misc_path = mne.datasets.misc.data_path()

raw = mne.io.read_raw(misc_path / 'ecog' / 'sample_ecog_ieeg.fif')
raw.info['bads'] = ['G10', 'G51']
raw.load_data()
raw.interpolate_bads()
```

### Expected results

Channels should be interpolated

### Actual results

```
Setting channel interpolation method to {}.
<ipython-input-10-fe35daf7ae7f>:2: RuntimeWarning: No bad channels to interpolate. Doing nothing...
```

### Additional information

I think for `ecog` the existing spherical interpolation is reasonable. Perhaps `seeg` should interpolate per lead but I think raising a `NotImplementedError` would be more helpful.

alexrockhill, 2024-01-03:
Tagging @mscheltienne @adam2392 @jasmainak for thoughts

adam2392, 2024-01-03:
> I think for `ecog` the existing spherical interpolation is reasonable. Perhaps `seeg` should interpolate per lead but I think raising a `NotImplementedError` would be more helpful.

I am +1 for both. Tho, perhaps the real challenge for us is showing the spherical interpolation is useful at all for users.

alexrockhill, 2024-01-03:
>  Tho, perhaps the real challenge for us is showing the spherical interpolation is useful at all for users.

In my particular use case there are a few pops and wayward sensors on an ECoG grid with very limited trials because it was intraoperative. I think in that case, tossing the whole epoch or the whole channel are not ideal so an interpolation seems like the best choice.

mscheltienne, 2024-01-04:
> I think for ecog the existing spherical interpolation is reasonable. Perhaps seeg should interpolate per lead

+1 for both, I never worked with intracranial data, do we store information about the grid and lead geometry?

Raising a `NotImplementedError` could be detrimental in an edge-case with a `Raw` containing multiple channel types that can be interpolated. For instance, combined stereo-EEG with scalp-EEG or with MEG recordings. I would however improve the `RuntimeWarning` to mention the channel types and bad channels that were *not* interpolated.

alexrockhill, 2024-01-04:
Maybe I'll just add a spline interpolation on an sEEG shaft at the same time, shouldn't be too hard.

closed 2024-01-10 by agramfort