mne-tools/mne-python #12344: Improving usage guidelines for running mne using wsl2

withmywoessner, 2024-01-10:
### Proposed documentation enhancement

Hello! I exclusively contribute to mne using wsl2 and was wondering if I could add a short section to the contribution guide that contains some of the steps I follow to make sure my environment works for mne development. Thanks!

larsoner, 2024-01-10:
In principle yes we want to make sure people can develop on all sorts of platforms/configs. However, are there MNE-specific things that were needed, or was it more (as I suspect) general WSL2<->GitHub type stuff that was needed? If it is more general than MNE, are there any existing tutorials/how-tos that we can link to instead of writing up our own?

hoechenberger, 2024-01-26:
We may also wish to describe to largely improved UX with GUI apps in WSL on Windows 11.

Additionally, I'll pick up work on the dev container setup again soon, which will hopefully make contributing much easier.