mne-tools/mne-python #12351: [BUG] `stc_near_sensors` unexpected behavior

alexrockhill, 2024-01-10:
### Description of the problem

Trying to use `stc_near_sensors` there is one bug if there are any bad channels, that causes 

```
if picks is None:
    picks = pick_types(evoked.info, ecog=True, seeg=True, dbs=True)
evoked = evoked.copy().pick(picks)
frames = set(evoked.info["chs"][pick]["coord_frame"] for pick in picks)
```

This to fail with an index error because when `evoked` gets picked the bad channels aren't included.

There is another bug/API confusion that the default for `surface` is `'pial`` so if you pass a volume source estimate and no argument for `surface` then you get an index error because it tried to use the pial surface which has the wrong indices for your volume source estimate. That seems unnecessarily confusing, I think it should work reasonably with default arguments for both surface and volume source estimates.

Also while doing a MWE, this link in the issue template is broken https://mne.tools/dev/overview/datasets_index.html.

### Steps to reproduce

```Python
import mne

misc_path = mne.datasets.misc.data_path()
subjects_dir = misc_path / "seeg"

raw = mne.io.read_raw(misc_path / "seeg" / "sample_seeg_ieeg.fif")
raw.info['bads'] = ['LENT 1']
trans = mne.coreg.estimate_head_mri_t("sample_seeg", subjects_dir)

epochs = mne.Epochs(raw, detrend=1, baseline=None)
epochs = epochs["Response"][0]  # just process one epoch of data for speed
evoked = epochs.average()

surface = subjects_dir / "sample_seeg" / 'bem' / 'inner_skull.surf'
vol_src = mne.setup_volume_source_space(
    subject="sample_seeg", subjects_dir=subjects_dir, surface=surface,
    pos=10)
vol_stc = mne.stc_near_sensors(
    evoked, trans, subject="sample_seeg",
    subjects_dir=subjects_dir, src=vol_src)  # surface=None)
```

### Expected results

Working stc

### Actual results

Two index errors

```
File ~/projects/mne-python/mne/source_estimate.py:3878, in <genexpr>(.0)
   3876     picks = pick_types(evoked.info, ecog=True, seeg=True, dbs=True)
   3877 evoked = evoked.copy().pick(picks)
-> 3878 frames = set(evoked.info["chs"][pick]["coord_frame"] for pick in picks)
   3879 if not frames == {FIFF.FIFFV_COORD_HEAD}:
   3880     raise RuntimeError(
   3881         "Channels must be in the head coordinate frame, " f"got {sorted(frames)}"
   3882     )

IndexError: list index out of range
```

```
File ~/projects/mne-python/mne/source_estimate.py:3930, in <listcomp>(.0)
   3928             rrs = apply_trans(trans, rrs)
   3929     else:
-> 3930         rrs = np.concatenate([s_rr[s["vertno"]] for s_rr, s in zip(surf_rr, src)])
   3931     keep_all = True
   3932 # ensure it's a usable one

IndexError: index 126967 is out of bounds for axis 0 with size 126967
```

### Additional information

None

alexrockhill, 2024-01-10:
cc @adam2392 @larsoner

alexrockhill, 2024-01-10:
Also when I use that to do `stc.plot_3d` it works the first time and then gets killed the second identical time it's called so I think there might be a memory leak but that's totally separate and probably an issue with `mne.viz.Brain`.

larsoner, 2024-01-11:
Yeah once it's picked you shouldn't use `picks` anymore so safest to `del picks` and just use `frames = set(ch["coord_frame"] for ch in evoked.info["chs"])`

closed 2024-02-08 by larsoner