mne-tools/mne-python #12355: Add `picks` argument to `Raw.plot()`

ivopascal, 2024-01-12:
### Describe the new feature or enhancement

`Raw.plot()` currently does not accept a `picks=` argument, even though `Evoked.plot()` and `Epochs.plot()` do.
For a mostly-unified API it would be nice to add this parameter.

### Describe your proposed implementation

This would be added as an argument to `Raw.plot()`. 
I suppose it can come after the `n_channels` argument. Ideally it would be the same for all classes, but that would break backward compatibility.

The behaviour and implementation should be similar to or identical to how it's done in `Epochs.plot()` and `Evoked.plot()`. 

### Describe possible alternatives

Alternatively, the entire function call could be rewritten to match Evoked and Epochs. The would be nicer for the future, but it would break a lot of things in the short term, for a relatively minor improvement.

### Additional context

Discussed on the MNE forum here: [Unexpected behaviour Raw.pick](https://mne.discourse.group/t/unexpected-behaviour-raw-pick/8132)

larsoner, 2024-01-12:
Technically [`order=pick_types(...)`](https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.plot) actually works like you suggest I think. But every time I want to do this I have to look up the argument name.

One backward-compatible way forward would be to do a legacy-style deprecation of `order`

1. put a `*` in the arg list for `raw.plot` before `order`
2. move `order=None` to the end
3. put `picks=None` where `order=None` was
4. allow `order` or `picks` to be specified, but not both

Eventually we could deprecate `order` but I suspect it's pretty widely used, and the backward compat code we'd need to keep around is only 5 lines I think:
```
if order is not None:
    if picks is not None:
        raise ValueError(...)
    picks = order
del order
```
then we can internally use `picks` instead of `order`, and process `picks` the normal way (`_picks_to_idx` internally will pass through the `array of int` that `order` currently supports!).

Let's see if others agree with this idea, but would you be interested in trying to make this change @ivopascal ?

ivopascal, 2024-01-12:
Yes, I'd be happy to implement it :)

drammock, 2024-01-15:
I'm fine with @larsoner's plan as long as `picks=[4, 3, 2, 1, 0]` works the same as `order[4, 3, 2, 1, 0]` (i.e., as long as it's still possible to put the channels in any arbitrary user-specified order on the plot)

ivopascal, 2024-01-24:
I said I'd implement it but I'm actually swamped with work right now, and it's taking me a bit more time than I expected to figure out how to work with the existing code.

I'd still be happy to implement it in March, but if someone else wants to put this together instead of waiting that's also good.

ivopascal, 2024-02-29:
Hmm, I don't think `picks` should replace `order`. `Epochs.plot()` and `Evoked.plot()` actually have both, which means you can have `picks` in an arbitrary order and preserve the "standard" ordering of channels. Only specifying `order` would then actually modify the order. 

I think having as-much-as-possible uniformity between `Epochs.plot()`, `Evoked.plot()` and `Raw.plot()` would be nicest, so I propose having `picks=[3,2,1]` to be different than `order=[3,2,1]` and we should then keep both parameters.

ivopascal, 2024-04-16:
Closing because #12467 is merged and fixes the issue.

closed 2024-04-16 by ivopascal