mne-tools/mne-python #12360: Replace percent format with format specifiers (f-strings)

cbrnr, 2024-01-15:
#12358 left a bunch of strings that need to be converted from percent formatting to ideally f-strings. Ruff rule UP031 can then be enabled in `pyproject.toml`.

cbrnr, 2024-01-26:
Great, thanks @Aditya3425-Fst! You can get a list of missing conversion by running `ruff check mne --config pyproject.toml --select UP031` in the MNE root directory. Currently, there are 359 errors:

<details>

```
mne/fixes.py:102:14: UP031 Use format specifiers instead of percent format
mne/fixes.py:193:21: UP031 Use format specifiers instead of percent format
mne/fixes.py:268:25: UP031 Use format specifiers instead of percent format
mne/fixes.py:278:25: UP031 Use format specifiers instead of percent format
mne/fixes.py:291:16: UP031 Use format specifiers instead of percent format
mne/forward/_compute_forward.py:664:15: UP031 Use format specifiers instead of percent format
mne/forward/_lead_dots.py:72:32: UP031 Use format specifiers instead of percent format
mne/forward/_lead_dots.py:77:32: UP031 Use format specifiers instead of percent format
mne/forward/_make_forward.py:302:17: UP031 Use format specifiers instead of percent format
mne/forward/_make_forward.py:696:21: UP031 Use format specifiers instead of percent format
mne/forward/forward.py:1260:17: UP031 Use format specifiers instead of percent format
mne/forward/forward.py:1270:13: UP031 Use format specifiers instead of percent format
mne/forward/forward.py:1422:17: UP031 Use format specifiers instead of percent format
mne/forward/forward.py:1427:17: UP031 Use format specifiers instead of percent format
mne/forward/forward.py:1494:21: UP031 Use format specifiers instead of percent format
mne/forward/forward.py:1516:13: UP031 Use format specifiers instead of percent format
mne/inverse_sparse/mxne_inverse.py:59:13: UP031 Use format specifiers instead of percent format
mne/inverse_sparse/mxne_inverse.py:815:13: UP031 Use format specifiers instead of percent format
mne/inverse_sparse/mxne_inverse.py:1093:25: UP031 Use format specifiers instead of percent format
mne/inverse_sparse/mxne_optim.py:246:30: UP031 Use format specifiers instead of percent format
mne/inverse_sparse/mxne_optim.py:528:29: UP031 Use format specifiers instead of percent format
mne/inverse_sparse/mxne_optim.py:548:18: UP031 Use format specifiers instead of percent format
mne/inverse_sparse/mxne_optim.py:643:13: UP031 Use format specifiers instead of percent format
mne/inverse_sparse/mxne_optim.py:1273:25: UP031 Use format specifiers instead of percent format
mne/inverse_sparse/mxne_optim.py:1506:13: UP031 Use format specifiers instead of percent format
mne/inverse_sparse/mxne_optim.py:1626:13: UP031 Use format specifiers instead of percent format
mne/io/artemis123/artemis123.py:134:47: UP031 Use format specifiers instead of percent format
mne/io/artemis123/tests/test_artemis123.py:39:30: UP031 Use format specifiers instead of percent format
mne/io/artemis123/tests/test_artemis123.py:43:32: UP031 Use format specifiers instead of percent format
mne/io/base.py:416:21: UP031 Use format specifiers instead of percent format
mne/io/base.py:1487:30: UP031 Use format specifiers instead of percent format
mne/io/base.py:1489:30: UP031 Use format specifiers instead of percent format
mne/io/base.py:1492:17: UP031 Use format specifiers instead of percent format
mne/io/base.py:1774:17: UP031 Use format specifiers instead of percent format
mne/io/base.py:2062:13: UP031 Use format specifiers instead of percent format
mne/io/base.py:2069:16: UP031 Use format specifiers instead of percent format
mne/io/base.py:2127:17: UP031 Use format specifiers instead of percent format
mne/io/base.py:2803:17: UP031 Use format specifiers instead of percent format
mne/io/brainvision/brainvision.py:867:21: UP031 Use format specifiers instead of percent format
mne/io/cnt/cnt.py:295:24: UP031 Use format specifiers instead of percent format
mne/io/ctf/ctf.py:230:17: UP031 Use format specifiers instead of percent format
mne/io/ctf/info.py:174:13: UP031 Use format specifiers instead of percent format
mne/io/ctf/info.py:220:21: UP031 Use format specifiers instead of percent format
mne/io/curry/curry.py:428:21: UP031 Use format specifiers instead of percent format
mne/io/egi/egimff.py:67:28: UP031 Use format specifiers instead of percent format
mne/io/egi/egimff.py:75:36: UP031 Use format specifiers instead of percent format
mne/io/egi/egimff.py:106:13: UP031 Use format specifiers instead of percent format
mne/io/egi/egimff.py:158:17: UP031 Use format specifiers instead of percent format
mne/io/eyelink/tests/test_eyelink.py:236:18: UP031 Use format specifiers instead of percent format
mne/io/eyelink/tests/test_eyelink.py:241:17: UP031 Use format specifiers instead of percent format
mne/io/eyelink/tests/test_eyelink.py:248:18: UP031 Use format specifiers instead of percent format
mne/io/fieldtrip/utils.py:56:17: UP031 Use format specifiers instead of percent format
mne/io/fieldtrip/utils.py:177:21: UP031 Use format specifiers instead of percent format
mne/io/fieldtrip/utils.py:366:28: UP031 Use format specifiers instead of percent format
mne/io/kit/coreg.py:89:15: UP031 Use format specifiers instead of percent format
mne/io/kit/coreg.py:166:17: UP031 Use format specifiers instead of percent format
mne/io/kit/coreg.py:172:13: UP031 Use format specifiers instead of percent format
mne/io/kit/kit.py:538:21: UP031 Use format specifiers instead of percent format
mne/io/nirx/nirx.py:104:33: UP031 Use format specifiers instead of percent format
mne/io/nirx/nirx.py:133:35: UP031 Use format specifiers instead of percent format
mne/io/nirx/nirx.py:146:36: UP031 Use format specifiers instead of percent format
[... cut ...]
mne/viz/utils.py:2154:25: UP031 Use format specifiers instead of percent format
mne/viz/utils.py:2215:17: UP031 Use format specifiers instead of percent format
mne/viz/utils.py:2323:21: UP031 Use format specifiers instead of percent format
```
</details>

Ruff does not find all occurences though, so in a second step, I'd do a global search for ` % ` (space percent space) and see what else crops up (I find 1617 results currently, but this likely includes the 359 occurences that Ruff reported).

Let me know if you have further questions!

Aditya3425-Fst, 2024-01-31:
@cbrnr  can you tell me the command for  passing the repository test on my local machine .

drammock, 2024-01-31:
> @cbrnr can you tell me the command for passing the repository test on my local machine .

the test for checking if you've fixed them all is the same command used for finding the errors:
`ruff check mne --config pyproject.toml --select UP031` (run from the root directory of your clone of the repository)

ughrima, 2024-02-02:
I've been working on resolving this issue, and each time I think I've fixed them, more errors seem to appear later on. This is causing some confusion, and I'm unsure about when it's the right time to push my work. Can someone please help me out?

cbrnr, 2024-02-02:
We currently have @Aditya3425-Fst working on this in #12405. If you want to contribute, please comment directly over there.

hasrat17, 2024-02-24:
Hey @cbrnr if this issue is still not resolved, then I would like to work on this. Can you assign it to me?

hasrat17, 2024-02-25:
Hi @cbrnr I have one doubt; in some places, we have code like this
`   name_pattern = "%s_div%%i%s" % (basename, name_ext)
    names = tuple(name_pattern % i for i in range(1, n_parts + 1))`
Here I am changing both the %s to f string like this `name_pattern = f"{basename}_div%%i{name_ext}"`
but for the next part, as we are using %%i I am using % only `names = tuple(name_pattern % i for i in range(1, n_parts + 1))`
So this is FINE, or do I need to change the code for using f string here also?

sappelhoff, 2024-07-27:
Re-opening, because even once #12758 is merged, there will still be many % strings to get rid of.

Please search globally for `" %` to find them.

scottrbrtsn, 2024-11-18:
Get all occurrences and line numbers with `grep -R '" %' -n --include=\*.py .`

Current count is 202 occurrances in 33 files: 
```
grep -R '" %' --include=\*.py . | wc -l
202

grep -R '" %' -l --include=\*.py . | wc -l
33
```

And a checklist: 
`grep -R '" %' -l --include=\*.py  .`

- [ ] ./mne/_fiff/tests/test_pick.py
- [ ] ./mne/channels/_dig_montage_utils.py
- [ ] ./mne/channels/layout.py
- [ ] ./mne/channels/montage.py
- [ ] ./mne/channels/tests/test_montage.py
- [ ] ./mne/cov.py
- [ ] ./mne/decoding/ssd.py
- [ ] ./mne/dipole.py
- [ ] ./mne/event.py
- [ ] ./mne/filter.py
- [ ] ./mne/forward/_field_interpolation.py
- [ ] ./mne/forward/forward.py
- [ ] ./mne/inverse_sparse/mxne_optim.py
- [ ] ./mne/io/base.py
- [ ] ./mne/io/brainvision/tests/test_brainvision.py
- [ ] ./mne/io/bti/bti.py
- [ ] ./mne/io/ctf/ctf.py
- [ ] ./mne/io/ctf/info.py
- [ ] ./mne/io/egi/egimff.py
- [ ] ./mne/io/fiff/tests/test_raw_fiff.py
- [ ] ./mne/io/kit/kit.py
- [ ] ./mne/io/kit/tests/test_coreg.py
- [ ] ./mne/label.py
- [ ] ./mne/minimum_norm/inverse.py
- [ ] ./mne/minimum_norm/tests/test_inverse.py
- [ ] ./mne/minimum_norm/time_frequency.py
- [ ] ./mne/preprocessing/ica.py
- [ ] ./mne/preprocessing/maxwell.py
- [ ] ./mne/rank.py
- [ ] ./mne/source_estimate.py
- [ ] ./mne/source_space/_source_space.py
- [ ] ./mne/surface.py
- [ ] ./mne/transforms.py

PR #12464 (2024-03-19): Fixes for #12360 Replacing percent format with f-strings

#### Reference issue
 FIXES #12360 

#### What does this implement/fix?
Replace percent format with format specifiers (f-strings)

#### Additional information
Used this command to capture percent format `ruff check mne --config pyproject.toml --select UP031`
[image]