mne-tools/mne-python #12372: MAINT: pip-pre failures

larsoner, 2024-01-19:
All PRs are failing pip-pre with for example:

https://github.com/mne-tools/mne-python/actions/runs/7562092509/job/20668202667?pr=12237

Looks like there is a lot of stuff to fix here including some invalid value and linalg stuff. :(

I'll try to look Monday

scott-huberty, 2024-01-22:
Yet More CI failures! 🤠  Looks like Pandas is warning of an additional upcoming hard dependency, causing a lot of MNE tests to fail, For example see https://github.com/mne-tools/mne-python/actions/runs/7618003617/job/20748304122?pr=12237 

and https://github.com/pandas-dev/pandas/issues/54466

Wasn't sure if this one was on your radar yet.

drammock, 2024-01-22:
> Wasn't sure if this one was on your radar yet.

Thanks for reporting. It showed up in the weekly CI job overnight, but I haven't done anything about it yet 😞

larsoner, 2024-01-23:
Yep fixes for that are in https://github.com/mne-tools/mne-python/pull/12377

closed 2024-01-29 by larsoner