mne-tools/mne-python #12373: Cortical signal suppression with magnetometers and EEG

jshanna100, 2024-01-21:
### Describe the new feature or enhancement

The [cortical signal suppression](https://pubmed.ncbi.nlm.nih.gov/30604048/) technique takes advantage of the differential sensitivities to subcortical sensitivities between channel types to filter out cortical signals. In the paper as well as in the already existing MNE-Python implementation, this is done with gradiometers and magnetometers. Going through the paper, however, I do not yet see any theoretical reason why it must necessarily be these two channel types. I would expect the same could be done with any two channel arrays with this differential sensitivity, and some amount of linear dependence, e.g. magnetometers and EEG. In the preliminary tests I have done, this does in fact seem to be the case; redoing the CSS example with mag and eeg channel types also produces a good result, though one must also reduce n_proj from 6 to 1 or 2.

Perhaps the functionality could be rewritten to offer the user more flexibility in applying this method? As it stands now, only systems with both magnetometers and gradiometers can take advantage of this very useful technique, which is a relatively limited amount of labs/datasets.

### Describe your proposed implementation

Rewrite the function such that the user can specify which two channel arrays to use, also maybe specifying some default behaviours, i.e. if left unspecified, checking which channel types are available and choosing appropriate ones, also changing the n_proj defaults accordingly (lower for mag+eeg, higher for grad+mag).

Some optional things:
1) Adding an option to return both the cortically filtered and subcortically filtered signal, in the event that someone actually wants to suppress the subcortical activity in the signal instead.
2) Expanding the example to show the added functionality.

### Describe possible alternatives

a

larsoner, 2024-01-22:
@johnsam7 @jasmainak any thoughts on this?

jasmainak, 2024-01-26:
It sounds like a research question ... I don't have a good handle on how using another set of channel types might work @johnsam7 perhaps you can pitch in?

larsoner, 2024-02-06:
> Rewrite the function such that the user can specify which two channel arrays to use, also maybe specifying some default behaviours, i.e. if left unspecified, checking which channel types are [...]

Have you tried using `mag_picks="mag", grad_picks="eeg"` with the existing code? I *think* this would get you the behavior you want, assuming it actually translates these pick names properly. If it doesn't, then we can make it do so pretty easily with `_picks_to_idx` I think. Then we can update the docs to state that `mag_picks` and `grad_picks` are really used to define two channel sets for which to find a joint subspace. (And we should update the docstring of `picks` to make it clear that it defines the channels that the projection should actually be applied to, which is not currently clear!)

In principle we could deprecated `grad_picks=None, mag_picks=None` in favor of something like `picks_a="grad", picks_b="mag"` or something and say it finds the joint subspace between the `a` and `b` channels, but not sure it's worth it.

larsoner, 2024-02-06:
@jshanna100 if you have time feel free to investigate further and open a PR directly, considering my question about trying it as rhetorical, to be answered by the PR based on whether or not you add a `_picks_to_idx` :slightly_smiling_face:

PR #12726 (2024-07-17): DOC: Document CSS more completely
Closes #12373

Turns out we do already have the flexibility / functionality needed, so this is a tiny test tweak (adding a second case) and a doc update so no changelog update needed.
closed 2024-07-17 by larsoner