mne-tools/mne-python #12379: Missing documentation for onset, duration, description attributes in Annotations class?

scott-huberty, 2024-01-22:
### Proposed documentation enhancement

A colleague of mine who is a bit of an MNE power user had no idea that you can do something like `raw.annotations.onset` to retrieve an array of the onsets of all individual annotations:

```
import mne

annot = mne.Annotations(onset=[1, 3, 5], duration=[1, 1, 1], description=['a', 'b', 'c'])
annot.onset 
# array([1., 3., 5.])
```

 As far as I can tell we don't document the `onset`, `duration`, `description` attributes in the [API documentation](https://mne.tools/dev/generated/mne.Annotations.html). Should we?

larsoner, 2024-01-23:
Yeah we could add them. I think we get them for free (maybe?) if we do:
```
def __init__(self, ...)
    ...
    self._onset, self._duration, ... = ...

@property
def onset(self):
    """..."""
    return self._onset

@onset.setter
def onset(self, onset)
    self._onset[:] = onset
```
Or something like this

scott-huberty, 2024-01-23:
Ok I'll test it out and open a PR when I have some time.

hoechenberger, 2024-01-24:
(Still thinking these should all be pluralized but this is a different discussion I suppose)