mne-tools/mne-python #12384: ENH: Add EEG support to find_bad_channels_maxwell

larsoner, 2024-01-24:
`find_bad_channels_maxwell` uses the SSS basis to find MEG channels that are bad. In principle the same thing should be doable for EEG using the [scalar spherical harmonic basis](https://pubmed.ncbi.nlm.nih.gov/32746058/). Code-wise I think the changes would be fairly simple, and it's a pretty straightforward extension of the existing bad-channel method conceptually. So I'm thinking it's worth implementing it and testing it on some data I have locally and in MNE-BIDS-Pipeline to see if it works reasonably well. I could compare to `autoreject` as well as [LOF](https://github.com/mne-tools/mne-python/pull/11234). Thoughts?

I'm thinking this could be implemented via a `find_bad_channels_maxwell(..., *, ch_type="meg" (default) | "eeg")` for example to allow finding bad channels for MEG or EEG channels. An alternative would be to add a `picks` argument but really you fit/process all channels of a given type jointly, so I tend toward liking `ch_type` instead. And I think it's okay to have to call it twice if you have both channel types, though in theory we could allow a `ch_type="all"` or `picks=("meg", "eeg")` or whatever to mean "loop over MEG and EEG channels), which would be okay, too.

@hoechenberger do you think this would then also be useful for MNE-BIDS(-Pipeline)?

sappelhoff, 2024-01-24:
I would also suggest comparing against pyprep's: [`pyprep.NoisyChannels(raw).find_all_bads()`](https://pyprep.readthedocs.io/en/latest/generated/pyprep.NoisyChannels.html#pyprep.NoisyChannels)

hoechenberger, 2024-01-24:
I have to say that find_all_Bads marks way too many channels as bad in my datasets. I usually only use a subset of the functions in pyprep (I can share which those are specifically in case you're interested)

hoechenberger, 2024-01-24:
Yes I believe it's usually the SNR criterium that fails for me.

agramfort, 2024-01-24:
It’s published works so I am supportive assuming it’s properly validated against what we already have or what pyprep has.