mne-tools/mne-python #12386: Excessive memory usage of `mne.io.read_raw_edf()`?

cbrnr, 2024-01-25:
### Description of the problem

It looks like `mne.io.read_raw_edf()` might be using too much memory, at least if I interpret the output of Memray correctly. For example, loading [this 163MB EDF file](https://osf.io/download/jasgc/) with `preload=False` consumes 691MB of RAM. With `preload=True`, the process requires 1.28GB. I've compared this to `edfio.read_edf()`, which uses only 319MB to read the file (corresponding to `preload=True`).

### Steps to reproduce

```Python
0. Install [Memray](https://github.com/bloomberg/memray) using `pip install memray`
1. Save the following code in a file called `mwe.py`:
   
   import mne
   
   raw = mne.io.read_raw_edf("example_nl22_edfc.edf", preload=True)
   
2. Run `memray run mwe.py`
3. Run `memray tree memray-mwe.py.XXXXX.bin` (replace `XXXXX` with the actual number in the file name)
```

### Link to data

https://osf.io/download/jasgc/

### Expected results

Memory consumption is not much higher than the file size (e.g. around 320MB like `edfio.read_edf()`).

### Actual results

Memory consumption is 4x higher, and even for `preload=False` it is more than 2x higher than loading the entire file into memory with `edfio`.

### Additional information

Platform             Linux-6.7.1-arch1-1-x86_64-with-glibc2.38
Python               3.12.1 (main, Jan 24 2024, 20:39:08) [GCC 13.2.1 20230801]
Executable           /home/clemens/.pyenv/versions/3.12.1/envs/mne/bin/python3.12
CPU                   (8 cores)
Memory               15.4 GB

Core
├☑ mne               1.7.0.dev79+ge0af51dec (devel, latest release is 1.6.1)
├☑ numpy             1.26.3 (unknown linalg bindings (threadpoolctl module not found: No module named 'threadpoolctl'))
├☑ scipy             1.12.0
├☑ matplotlib        3.8.2 (backend=TkAgg)
├☑ pooch             1.8.0
└☑ jinja2            3.1.3

Numerical (optional)
└☐ unavailable       sklearn, numba, nibabel, nilearn, dipy, openmeeg, cupy, pandas

Visualization (optional)
└☐ unavailable       pyvista, pyvistaqt, vtk, qtpy, ipympl, pyqtgraph, mne-qt-browser, ipywidgets, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo

cbrnr, 2024-01-25:
Update: I should mention that the EDF file contains 193 channels with 421070 samples each, which equals 193 * 421070 = 81266510 float64 elements. 81266510 * 8 / 1024**2 ≈ 620MB.

drammock, 2024-01-25:
Sounds like edfio is using lower precision? is that enough of an explanation here, or do you want to keep digging?

cbrnr, 2024-01-25:
No, `edfio` is also using double precision. Maybe something is off with the memray output, or maybe I'm reading it incorrectly. Don't you think it is odd that we use the memory required to hold the entire data when `preload=False`? And twice that amount for `preload=True`?

drammock, 2024-01-25:
> Don't you think it is odd that we use the memory required to hold the entire data when `preload=False`? And twice that amount for `preload=True`?

yes I do. Sorry, my last reply was from a phone, I lost track of the fact that 691MB was the `preload=False` case. That is indeed weird / worth digging into.

drammock, 2024-01-25:
> No, `edfio` is also using double precision.

But wait, I thought you said `edfio` corresponds to the `preload=True` case?  I think you've said each of these things, and it seems like at least one of them must be wrong:

- edfio loads data as double-precision floats
- edfio doesn't memmap the data (AKA loads it into RAM)
- the EDF file contains 193 channels with 421,070 samples each, which equals 193 * 421,070 = 81,266,510 float64 elements. 81,266,510 * 8 / 1024**2 ≈ 620MB.
- edfio uses 319MB to load this file

cbrnr, 2024-01-25:
Yeah, the last point must be incorrect, but that's what memray told me. I was probably misinterpreting the output, or maybe I screwed up the profiling somehow. I'll try to replicate.

cbrnr, 2024-01-26:
I tried another memory profiler, this time `memory-profiler`, but I seem to get the same strange results:

```python
from time import sleep

from edfio import read_edf

from mne.io import read_raw_edf

fpath = "/Users/clemens/Data/biosignal-test-data/EDF/example_nl22_edfc.edf"

edf = read_edf(fpath)
sleep(2)

raw = read_raw_edf(fpath, preload=False)
sleep(2)

raw = read_raw_edf(fpath, preload=True)
sleep(2)

print(raw._data.nbytes / 1024**2)
```

Running this script (saved as `mwe.py`) with

```
mprof run mwe.py
```

and then

```
mprof plot
```

produces this plot:

[image]

and output

```
620.0142669677734
```

So I think I must be doing something completely wrong, because why is `edfio` using 320MB? All number don't make sense...

closed 2024-01-26 by cbrnr

cbrnr, 2024-01-26:
FWIW, the reason why `edfio` uses less memory is because it loads the `int16` signals (as stored in an EDF file). It only converts to `float64` upon request (when accessing the signals).

drammock, 2024-01-26:
Thanks for following up. Loading as int16 would be a big change for us that would be inconsistent with what we do elsewhere, so I'm afraid we may never match edfio mem usage

hoechenberger, 2024-01-29:
Nice detective work!! 🕵️