mne-tools/mne-python #12387: Consider removing jinja2 from core dependencies

cbrnr, 2024-01-25:
We've had many discussions on how we should handle dependencies, but given that we currently only include absolutely essential packages as core dependencies, I was wondering why we include `jinja2`. In my opinion, it would be better (more consistent) to make it an optional dependency. In addition, it is only imported in one location (`mne/html_templates/_templates.py`), and even there it is a nested import.

drammock, 2024-01-25:
I'm pretty sure this has been discussed before, can you provide a cross ref to the past discussion(s)?

cbrnr, 2024-01-25:
😄 yes, it was me who asked this question before in #11559. Apparently, it still bothers me, but I guess your opinions won't have changed, so feel free to close.

cbrnr, 2024-01-26:
I think I know what bothers me: listing `jinja2` and `pooch` in the "Core" section of the output of `mne.sys_info()`. This doesn't really make sense, so what about removing these two packages from the section? We don't include `lazy-loader` either (which makes sense).

hoechenberger, 2024-01-26:
Good idea to make the core deps section smaller by narrowing it down to the essentials  👍

drammock, 2024-01-26:
I'm open to that. Where should they go instead if not in "core"?

-------- Original Message --------

cbrnr, 2024-01-26:
Nowhere, or to a new section called "infrastructure" or something like that.

PR #12411 (2024-02-02): Remove `jinja2` and `pooch` from `sys_info`
Fixes #12387.
closed 2024-02-02 by larsoner