mne-tools/mne-python #12388: Register mne-python with scientific-python for lazy_loader adoption

bjlittle, 2024-01-25:
### Proposed documentation enhancement

@larsoner et al

Just thought I'd suggest that you guys add yourselves to https://github.com/scientific-python/lazy_loader/issues/50, since you're leveraging the awesome benefits of the `lazy_loader` package :

drammock, 2024-01-25:
> Just thought I'd suggest that you guys add yourselves to [scientific-python/lazy_loader#50](https://github.com/scientific-python/lazy_loader/issues/50)

Sure, feel free to add us there if it's helpful to the lazy loader devs to do so. IIRC we're already listed on the relevant SPEC website as an adopter.

cbrnr, 2024-01-25:
To be honest, I'm really not convinced that the benefits of this package outweigh its issues. The adoption has been controversial, and I specifically don't like to further encourage other projects to adopt it without any specific reason.

hoechenberger, 2024-01-25:
I'm not sure why this would capture anyone's interest 😅 Since MNE-Python isn't listed as a [core project](https://scientific-python.org/specs/core-projects/#core-projects), I'm puzzled about the significance of understanding whether it employs lazy-loading or not, except perhaps for the users of MNE-Python itself.

Not veto-ing an addition to the list; I just don't think it serves any useful purpose.

drammock, 2024-01-25:
@hoechenberger and @cbrnr did you look at the issue that @bjlittle linked to? Its purpose is for lazy loader devs to know who to check with if they are considering API changes. I doubt it will have the effect of encouraging others to adopt it

bjlittle, 2024-01-25:
Thanks guys. It's genuinely interesting to hear the back story here to `lazy_loader` and your experiences.

If you don't mind, can you summarise the controversy on `mne-python` surrounding adoption? Even, your general personal objections, or not, would be super useful to know.

Many thanks 👍

drammock, 2024-01-25:
@bjlittle we started with #11838 (but did it a dumb way, without `.pyi` files). This caused problems with IDE completion hints etc, as discussed in #12059 (that's where all the opinions are aired).  The IDE problem was fixed by #12072.

hoechenberger, 2024-01-26:
@drammock Indeed I misunderstood the purpose of this issue – I thought it's to gather a list of projects to be publicized on the website! My bad.

cbrnr, 2025-02-17:
I just found another problem with lazy loading, which prevents me from creating a standalone installer: https://groups.google.com/g/pyinstaller/c/hzY32N5lZGA (see also https://github.com/cbrnr/mnelab/issues/294)

cbrnr, 2025-02-19:
Yes, there is no such file. In fact, the entire `mne` directory is missing. It turns out that the `--collect-all mne` argument fixes this issue!