mne-tools/mne-python #12396: ENH: Add param to report.add_epochs to report reject/flat

larsoner, 2024-01-30:
> I think the value of the "reject" parameter should be included via upstream changes to mne.Report.add_epochs().

_Originally posted by @hoechenberger in https://github.com/mne-tools/mne-bids-pipeline/issues/839#issuecomment-1917498418_

In other words, `epochs.reject` and maybe `epochs.flat` could be useful to allow opt-in to reporting in `add_epochs` so people know what rejection has been done. See linked PR for how it's done in MNE-BIDS-Pipeline for example.

Forchapeatl, 2024-02-24:
Hello @larsoner , I am Forcha Pearl from Cameroon. It's a pleasure meeting you.  I would love to participate in GSOC 2024 under mne-tools, Please is this what you mean ?

```diff
    @fill_doc
    def add_epochs(
        self,
        epochs,
        title,
        *,
        psd=True,
        projs=None,
        image_kwargs=None,
        topomap_kwargs=None,
        drop_log_ignore=("IGNORED",),
        tags=("epochs",),
        replace=False,
+       reject=epochs.reject
    ):
```

larsoner, 2024-02-26:
Not exactly -- I actually don't think we need any new parameter, but should just automatically always add a subsection that is basically the contents of this:

https://github.com/mne-tools/mne-bids-pipeline/blob/61b35530b16a61753b25fa663b575d9b2f7a6ed5/mne_bids_pipeline/steps/preprocessing/_09_ptp_reject.py#L215-L221

So just a new `add_html` in mne.Report._add_epochs, probably inside this conditional:

https://github.com/mne-tools/mne-python/blob/ae6e55e8eb41f080d6c209c129bb86ecdb2fc30d/mne/report/report.py#L3939

Probably need one for `reject` and one for `flat`.

larsoner, 2024-03-06:
Something like that. Not sure what `kind` is in your example -- that might be MNE-BIDS-Pipeline specific. Maybe try implementing something and running the report tutorial and see if it works as expected?

positiveonly, 2025-03-29:
Hi @larsoner ,

I am interested in working on this issue! I have gone through the discussion and understand that the goal is to modify Report.add_epochs() to include rejection and flat threshold details automatically. I will follow the approach of adding an HTML subsection with add_html(), similar to the MNE-BIDS-Pipeline implementation.

Before proceeding, I’d appreciate any additional guidance or suggestions you might have. Let me know if there's anything specific I should keep in mind while implementing this.

Thanks! Looking forward to contributing. 🚀

sruthisureshkumar-arch, 2026-10-23:
Hi @larsoner, I'd like to work on this issue! I'm thinking of automatically adding a table of the reject and flat thresholds in `_add_epochs` inside `if epochs._bad_dropped:` as you suggested but with readable units (e.g µV) ! 
p.s this picks up from #13617 which was closed. I'll be opening up a PR of my own targeting this issue!