mne-tools/mne-python #12398: MNE Epochs notch_filter feature request (re: #12092)

ikarabinas, 2024-01-30:
### Describe the new feature or enhancement

Currently, notch_filter can be applied to raw objects but does not have the functionality to filter epochs, as brought up previously by user hyphenzhao in request #12092. With the way I set up my pipeline, I epoch the data early on to apply other cleaning steps and am intentionally applying the notch filter at a later step. It would be very useful to be able to apply notch_filter to epoched data, in my case to avoid converting epochs back into a raw object in order to apply the notch filter. 

### Describe your proposed implementation

User hyphenzhao described this nicely in their post. Briefly, my understanding is that each epoch could be treated as a separate record, and the width of the notch filter could be adjusted partly based on the epoch length. 

### Describe possible alternatives

The alternative is to reconstruct a raw object from the epoched data by concatenating the epochs into a continuous numpy array, then using mne.io.RawArray to create a new raw object. This is what I am doing now in order to use the notch filter, but I'm worried it may introduce edge artifacts. How would you recommend handling possible edge artifacts introduced by the conversion back into a raw object?

larsoner, 2024-01-31:
@ikarabinas in principle I think this should be doable by moving `notch_filter` from this class:

https://github.com/mne-tools/mne-python/blob/195a2cc8009160fd125e355b0280e903a941c874/mne/io/base.py#L1147

to this class

https://github.com/mne-tools/mne-python/blob/195a2cc8009160fd125e355b0280e903a941c874/mne/filter.py#L2459

Would you be up for trying a PR to make this change?

ikarabinas, 2024-01-31:
@larsoner Great! I hope this works. Let me know if the edits I made in my PR reflect the changes you were referring to.