mne-tools/mne-python #12407: 'custom_ref_applied' is True event if the offline rereferencing is an average

nabilalibou, 2024-02-01:
### Proposed documentation enhancement

Hello,

contrary to what is written in the doc, `'custom_ref_applied'` is True event if the offline rereferencing is an average one.

`set_eeg_reference(ref_channels="average")` raises the flag `'custom_ref_applied'` even so it is said for this Attribute on https://mne.tools/stable/generated/mne.Info.html:

> custom_ref_applied:
>     Whether a custom (=other than average) reference has been applied to the EEG data. This flag is checked by some algorithms that require an average reference to be set.

I find it strange that this flag is raised by an 'average' reref (which rereferencing does not raise it then ?) but I am guessing that there are reasons about this. I would like to know the explanation and thinking behind why this case happens pls.
But in any case i think that this little doc paragraphe should be changed to reflect how it really works.
Since I'm already on this similar [issue](https://github.com/mne-tools/mne-python/issues/12401) I volunteer to continue to spot and correct the little contradictions around rereferencing and projections in the doc.

larsoner, 2024-02-02:
Yeah it should say "a custom (=other than average projector)` or similar. It's for inverse modeling reasons, we need to make sure an average reference over the remaining good channels is applied. Adding an average reference projector as the reference is a safe way to do this.

> I find it strange that this flag is raised by an 'average' reref (which rereferencing does not raise it then ?) but I am guessing that there are reasons about this. if anyone could enlighten me as to the reason behind it, I'd be grateful :).

Let's discuss this over in #12369.

nabilalibou, 2024-02-02:
Ok thanks for the link , I'll follow the discussion https://github.com/mne-tools/mne-python/issues/12369.
I'll open a PR to add this detail on the doc

larsoner, 2024-02-07:
I think this was taken care of in #12420 and #12426

PR #12426 (2024-02-07): DOC: Specifies that a custom reference is anything but an average projector
Fixes #12407

Label `no-changelog-entry-needed` to add i think
closed 2024-02-07 by larsoner