mne-tools/mne-python #12409: Proposal for Enhanced EEG Dataset Comparison Across Different Montages

incubodac, 2024-02-01:
### Describe the new feature or enhancement

Dear MNE Community,
I hope this message finds you well. I am currently working with two EEG datasets stemming from the same experimental paradigm, acquired using different caps (biosemi128/64). During my analysis, I encountered a challenge in facilitating a spatial downsampling of the 128-channel montage to enable a meaningful comparison of results between both datasets.
Given that only 34 electrode locations coincide between the two montages, I devised a solution by creating a virtual montage. This involved incorporating the 30 non-matching locations onto the 128-channel head, utilizing spline interpolation to obtain data for these additional points, and subsequently discarding the unwanted channels, then resulting  in a 64-channel montage dataset.
While my specific case addresses only biosemi caps, I believe this approach could be valuable in a broader context. It may not be uncommon to compare EEG datasets acquired using different montages. Therefore, I propose the inclusion of a more generalized version of this downsampling method to accommodate various montages.However, perhaps the effectiveness of this enhancement may be limited to montages with electrode locations calculated from a sphere, like the biosemi setup.
I welcome feedback and collaboration from the community to refine and implement this proposed feature.

### Describe your proposed implementation

Maybe as a method of DigMontage class.

### Describe possible alternatives

Not sure.

agramfort, 2024-02-01:
you could start by making an example doing this?

> Message ID: ***@***.***>
>

incubodac, 2024-02-15:
Yes, I've already prepared an example utilizing a BioSemi montage. You can
find it here <https://github.com/incubodac/Cross-MontageComparison>. I'm
eager to hear any feedback or suggestions you may have regarding it. Thank
you!

agramfort, 2024-02-16:
I see. You could make a new file in
https://github.com/mne-tools/mne-python/tree/main/examples/preprocessing
that is a full example.

sappelhoff, 2025-04-17:
- closed in https://github.com/mne-tools/mne-python/pull/13044

closed 2025-04-17 by sappelhoff