mne-tools/mne-python #12425: Annotation-related error reading EEGLAB file

cbrnr, 2024-02-07:
### Description of the problem

When trying to read a specific EEGLAB file, there is an error related to annotations and absolute onset time. This problem was reported in our forum: https://mne.discourse.group/t/read-raw-eeglab-gives-valueerror-cannot-convert-float-nan-to-integer/8285

### Steps to reproduce

```Python
import openneuro as on

import mne

on.download(
    dataset="ds004842",
    include=[
        "sub-001/**/*run-1*.set",
        "sub-001/**/*run-1*.fdt",
    ],
)

raw = mne.io.read_raw_eeglab(
    "ds004842/sub-001/ses-C1C/eeg/sub-001_ses-C1C_task-DriveOnMission_run-1_eeg.set"
)
```

### Link to data

https://openneuro.org/datasets/ds004842/versions/1.0.0

### Expected results

I should be able to import the data.

### Actual results

There is an error when importing the data:

```
<stdin>:1: RuntimeWarning: Unknown types found, setting as type EEG:
inferiorveog: ['EXG4']
leftheog: ['EXG1']
leftmastoid: ['EXG5']
rightheog: ['EXG2']
rightmastoid: ['EXG6']
superiorveog: ['EXG3']
Reading /Users/clemens/Projects/mne-python/ds004842/sub-001/ses-C1C/eeg/sub-001_ses-C1C_task-DriveOnMission_run-1_eeg.fdt
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/clemens/Projects/mne-python/mne/io/eeglab/eeglab.py", line 332, in read_raw_eeglab
    return RawEEGLAB(
           ^^^^^^^^^^
  File "<decorator-gen-238>", line 12, in __init__
  File "/Users/clemens/Projects/mne-python/mne/io/eeglab/eeglab.py", line 505, in __init__
    self.set_annotations(annot)
  File "<decorator-gen-225>", line 12, in set_annotations
  File "/Users/clemens/Projects/mne-python/mne/io/base.py", line 734, in set_annotations
    new_annotations.crop(
  File "<decorator-gen-38>", line 12, in crop
  File "/Users/clemens/Projects/mne-python/mne/annotations.py", line 636, in crop
    absolute_onset = timedelta(seconds=onset) + offset
                     ^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: cannot convert float NaN to integer
```

### Additional information

```
Platform             macOS-14.3-arm64-arm-64bit
Python               3.12.1 (main, Jan 24 2024, 11:32:51) [Clang 15.0.0 (clang-1500.1.0.2.5)]
Executable           /Users/clemens/.pyenv/versions/3.12.1/envs/mne/bin/python3.12
CPU                  arm (12 cores)
Memory               32.0 GB

Core
├☑ mne               1.7.0.dev77+g566c6ea5a (devel, latest release is 1.6.1)
├☑ numpy             1.26.3 (OpenBLAS 0.3.23.dev with 12 threads)
├☑ scipy             1.12.0
└☑ matplotlib        3.8.2 (backend=MacOSX)

Numerical (optional)
└☐ unavailable       sklearn, numba, nibabel, nilearn, dipy, openmeeg, cupy, pandas

Visualization (optional)
├☑ qtpy              2.4.1 (PySide6=6.6.1)
├☑ pyqtgraph         0.13.3
├☑ mne-qt-browser    0.6.1
└☐ unavailable       pyvista, pyvistaqt, vtk, ipympl, ipywidgets, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo
```

withmywoessner, 2024-02-07:
Hello @cbrnr I was actually about to make an issue for this. I can try to work on this soon. I imported it into EEGLAB and I believe it just ignores them:
[image]

cbrnr, 2024-03-01:
@withmywoessner are you still up for fixing this issue? Just asking if we can try to get it into the next release. If not, that's totally fine, and we'll just move it to a future release.

withmywoessner, 2024-03-01:
Hey @cbrnr! I got a little busy with some other things and forgot to make a PR. I'll make one this weekend. Is 1.7 planned on being released on April 14th?

cbrnr, 2024-03-01:
No worries! The milestone deadline is set to April 15, so I guess this is what we're aiming for (but @larsoner can tell for sure).

larsoner, 2024-03-01:
Yeah mid April is the tentative plan

PR #12484 (2024-03-13): [BUG] Drop annotations with NaN onset in EEGLAB raw files 
#### Reference issue
Fixes #12425.

#### What does this implement/fix?
Fix bug where MNE cannot handle EEGLAB events with NaN annotations

#### Additional information
Rationale for decision from EEGLAB documentation:
[image]
https://eeglab.org/tutorials/04_Import/Importing_Event_Epoch_Info.html
closed 2024-03-13 by larsoner