mne-tools/mne-python #12427: Decorators in traceback?

cbrnr, 2024-02-07:
I noticed that tracebacks can contain a lot of lines like `File "<decorator-gen-238>", line 12, in __init__`, as demonstrated e.g. in #12425:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/clemens/Projects/mne-python/mne/io/eeglab/eeglab.py", line 332, in read_raw_eeglab
    return RawEEGLAB(
           ^^^^^^^^^^
  File "<decorator-gen-238>", line 12, in __init__
  File "/Users/clemens/Projects/mne-python/mne/io/eeglab/eeglab.py", line 505, in __init__
    self.set_annotations(annot)
  File "<decorator-gen-225>", line 12, in set_annotations
  File "/Users/clemens/Projects/mne-python/mne/io/base.py", line 734, in set_annotations
    new_annotations.crop(
  File "<decorator-gen-38>", line 12, in crop
  File "/Users/clemens/Projects/mne-python/mne/annotations.py", line 636, in crop
    absolute_onset = timedelta(seconds=onset) + offset
                     ^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: cannot convert float NaN to integer
```

This has cropped up for some time now (probably a long time), but I've only recently noticed it. Since these lines are rather distracting, I was wondering if there was a way to suppress this output.

larsoner, 2024-02-07:
I've tried a couple of times to figure out a way to remove them and never did find a way

cbrnr, 2024-02-07:
It seems like you shouldn't really do this (for obvious reasons), but since almost all our functions are decorated, it gets a little out of control in the traceback. Since this is mainly the `verbose` decorator, I was wondering if there was a decorator-less alternative to achieve the same thing. It's probably not worth it, but out of curiosity, we're "just" adding behavior to an existing `verbose` parameter, right?

closed 2024-02-09 by cbrnr