mne-tools/mne-python #12458: ENH: Add tutorial on narrowband filters using real data

alexrockhill, 2024-02-23:
### Describe the new feature or enhancement

I think it would be nice to have an example or tutorial looking at filters on real data in order to look at task-based differences, not just preprocessing.

### Describe your proposed implementation

I think people might really benefit from seeing FIR vs IIR and the parameters and their effect on something well documented like movement-related beta in the bci dataset.

### Describe possible alternatives

There are other datasets, maybe visual alpha or an auditory biomarker that could make sense.

agramfort, 2024-02-24:
maybe a great blog post first?

> Message ID: ***@***.***>
>

alexrockhill, 2024-02-26:
Sounds good, will link to it in this issue and then we can decide whether to add it to documentation

closed 2024-02-29 by alexrockhill