mne-tools/mne-python #12465: RuntimeError: Incorrect number of samples (0 != 49999989), please report this error to MNE-Python developers

XuefengYang1128, 2024-02-27:
### Description of the problem

I can successfully read my data using the code:
raw = mne.io.read_raw_brainvision(EEG_PATH, verbose=True);
However, this code can not be executed successfully:
data = raw.get_data(units='uV')

### Steps to reproduce

```Python
raw = mne.io.read_raw_brainvision(EEG_PATH, verbose=True);
data = raw.get_data(units='uV')
```

### Expected results

data matrix

### Actual results

[image]
[image]
[image]

### Additional information

Platform             Windows-10-10.0.22631-SP0
Python               3.11.5 | packaged by Anaconda, Inc. | (main, Sep 11 2023, 13:26:23) [MSC v.1916 64 bit (AMD64)]
Executable           C:\Users\yangx\anaconda3\python.exe
CPU                  AMD64 Family 25 Model 116 Stepping 1, AuthenticAMD (16 cores)
Memory               15.2 GB

larsoner, 2024-02-29:
@XuefengYang1128 was there just some bug with your file?

XuefengYang1128, 2024-03-01:
could be, when I moved the data from the server to local area, the bug no longer exist.