mne-tools/mne-python #12466: The unit of the "Spectral" subplot in the ICA properties figure

hoechenberger, 2024-02-27:
When calling `ICA.plot_properties()`, the ordinate of the subplot titled "Spectrum" is labeled "AU^2/Hz (dB)":

[image]

(taken from https://mne.tools/stable/auto_tutorials/preprocessing/40_artifact_correction_ica.html#looking-at-the-ica-solution)

- Why "AU" and not the unit of the sensor type (e.g., µV for EEG)?
- dB here means – what? That the plotted value is 10 log10 (AU^2/Hz)?

A comparable figure produced by EEGLAB looks like this:

[image]

(taken from https://eeglab.org/tutorials/08_Plot_data/Plotting_ICA_components.html)

I feel that the axis label makes it much clearer to understand what's being shown here.

cbrnr, 2024-02-28:
I think we forgot to change this in #12232, where we converged on using:

Power (dB µV²/Hz)

Regarding the title, I'm OK with changing it to PSD if you think this is better.

hoechenberger, 2024-02-28:
This sounds reasonable! I don't have strong feelings regarding the plot title, I just thought the current wording might not be the best choice