mne-tools/mne-python #12485: Bug about rename_channels after add_channels

RunKZhang, 2024-03-07:
### Description of the problem

Hello, this this Runkai. I recently found a problem with applying raw.rename_channels to newly added channels. 

I use raw.add_chennels to add to new channels called 'PZ’ and 'FZ' for consistency in number of channels (you know in machine learning research, it is often to keep the number of channels same). After adding the two channels, I want to rename them to 'Pz' and 'Fz' using raw.rename_channels. However, there is a Keyerror occured. After checking the source code, I found it is because raw._orig_units do not have 'PZ' and 'FZ' in their keys. Finally, I fix the problem with directly assignment values to raw._orig_units.

So, my questions are as follows:
1. does "renaming operation" to newly added channel not supported?
2. if _orig_units is protected and should not be changed after initializatoin?

### Steps to reproduce

```Python
tmp_arr = np.zeros([len(chns_need_added), len(raw)])
tmp_info = mne.create_info(ch_names=chns_need_added， sfreq=raw.info['sfreq'], ch_types='eeg)
tmp_raw = mne.io.RawArray(tmp_arr, tmp_info)
raw = raw.add_channels([tmp_raw], force_update_info=True)

# error occur
channel_renaming_dict = {'PZ':'Pz', 'FZ':'Fz'}
raw.rename_channels(channel_renaming_dict)

# fix
for ch in chns_need_added:
  if ch not in raw._orig_units.keys():
     raw._orig_units[ch] = 'uV'
channel_renaming_dict = {'PZ':'Pz', 'FZ':'Fz'}
raw.rename_channels(channel_renaming_dict)
```

### Expected results

I think rename to newly_added_chns should be supported.

### Actual results

KeyError: 'PZ'

### Additional information

mne: 1.0.3

RunKZhang, 2024-03-08:
Hello，I tried to fix the bug today. And I found it is fixed and no error occur in a newer version. I came across the problem because I was using a older version. So, I think it is not a bug now.

Also, I think `_orig_units` should not be changed during processing, because it is directly read from header files. Thus, for added channels from `RawArray`, they have no `_orig_units`.

mscheltienne, 2024-03-10:
Can not reproduced on `main` with:

```
import numpy as np
import mne

raw = mne.io.RawArray(np.zeros([5, 1000]), mne.create_info(5, 1000, ch_types="eeg"))
chns_need_added = ["PZ", "FZ"]
tmp_arr = np.zeros([len(chns_need_added), len(raw)])
tmp_info = mne.create_info(
    ch_names=chns_need_added, sfreq=raw.info["sfreq"], ch_types="eeg"
)
tmp_raw = mne.io.RawArray(tmp_arr, tmp_info)
raw = raw.add_channels([tmp_raw], force_update_info=True)

# error occur
raw.rename_channels({"PZ": "Pz", "FZ": "Fz"})
```

Or with:

```
import numpy as np
import mne

raw = mne.io.RawArray(np.zeros([5, 1000]), mne.create_info(5, 1000, ch_types="eeg"))
for ch in raw.ch_names:
    raw._orig_units[ch] = "uV"
chns_need_added = ["PZ", "FZ"]
tmp_arr = np.zeros([len(chns_need_added), len(raw)])
tmp_info = mne.create_info(
    ch_names=chns_need_added, sfreq=raw.info["sfreq"], ch_types="eeg"
)
tmp_raw = mne.io.RawArray(tmp_arr, tmp_info)
raw = raw.add_channels([tmp_raw], force_update_info=True)

# error occur
raw.rename_channels({"PZ": "Pz", "FZ": "Fz"})
```

As you were using 1.0.3 and also report the bug as fix, I'll close.

closed 2024-03-10 by mscheltienne