mne-tools/mne-python #12486: Add method for channel interpolation to predefined channel positions/montage

apmellot, 2024-03-07:
### Describe the new feature or enhancement

I have recently been working with datasets with diverse montages and wanted to make them comparable by ensuring a consistent number and position of channels across datasets. Currently, achieving this requires hacking in MNE internal functions such as `mne.channels.interpolation._map_meg_or_eeg_channels` or `mne.channels.interpolation._make_interpolation_matrix`, and is not straightforward for users.
This new method would simplify the process. I believe a similar issue was raised in #12409.

### Describe your proposed implementation

The proposed implementation involves introducing a new method, `interpolate_to`, to either the `mne.io.Raw` or `mne.Epochs` class. This method would:
- Extract the existing channel positions from the input data.
- Create a new info file with the desired final channel positions defined by the provided montage or template.
- Use existing interpolation functions like `mne.channels.interpolation._map_meg_or_eeg_channels` or `mne.channels.interpolation._make_interpolation_matrix` to interpolate the existing channels to the desired positions.
A possible API for this method is:
`epoch.interpolate_to(montage: DigMontage, method: str, reg: float)`
where **montage** is the desired final montage, **method** is the interpolation method to use (either spherical spline or field interpolation), and **reg** is the regularization parameter to pass to the chosen interpolation function.

### Describe possible alternatives

The parameter **montage** could also be a list of str with the position names, or a str for the name of a template montage that would be defined in MNE.

apmellot, 2024-03-07:
@antoinecollas @agramfort

larsoner, 2024-03-07:
I think this can be tackled at the same time and using hopefully the same API as https://github.com/mne-tools/mne-python/issues/9609, can you check?

larsoner, 2024-03-07:
... specifically this proposal:

https://github.com/mne-tools/mne-python/issues/9609#issuecomment-901203391

agramfort, 2024-03-07:
+1 for interpolate_to method next to interpolate_bads

sappelhoff, 2024-07-27:
+1, this would be a cool feature

closed 2025-02-10 by larsoner