mne-tools/mne-python #12493: Resolution issue with EDF export

cbrnr, 2024-03-13:
### Description of the problem

When exporting [this BDF file](https://osf.io/download/5fhye/) to EDF, the amplitude resolution seems to decrease a lot (amplitude values are clearly discretized). I suspect that maybe this has to do with some physical to digital min/max setting, which by default is set to `"auto"`.

### Steps to reproduce

```Python
import mne

raw = mne.io.read_raw("S01.bdf")
raw.plot(duration=1, n_channels=1)  # plot 1
raw.export("S01.edf", overwrite=True)
raw_edf = mne.io.read_raw("S01.edf")
raw_edf.plot(duration=1, n_channels=1)  # plot 2
```

### Expected results

The data in the exported EDF file should be identical to the original BDF file.

### Actual results

This is a plot of a portion of the original BDF data (plot 1 in the MWE):
[image]
And here's the plot of the exported EDF file (plot 2 in the MWE):
[image]

### Additional information

```
Platform             macOS-14.3.1-arm64-arm-64bit
Python               3.12.2 (main, Feb 14 2024, 07:24:33) [Clang 15.0.0 (clang-1500.1.0.2.5)]
Executable           /Users/clemens/.pyenv/versions/3.12.2/envs/mne/bin/python
CPU                  arm (12 cores)
Memory               32.0 GB

Core
├☑ mne               1.7.0.dev26+g7bf1b4ab7 (devel, latest release is 1.6.1)
├☑ numpy             1.26.4 (OpenBLAS 0.3.23.dev with 12 threads)
├☑ scipy             1.12.0
└☑ matplotlib        3.8.3 (backend=MacOSX)

Numerical (optional)
├☑ sklearn           1.4.1.post1
└☐ unavailable       numba, nibabel, nilearn, dipy, openmeeg, cupy, pandas

Visualization (optional)
└☐ unavailable       pyvista, pyvistaqt, vtk, qtpy, ipympl, pyqtgraph, mne-qt-browser, ipywidgets, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo
```

cbrnr, 2024-03-14:
Thanks @Teuniz! I think this would be a good option, although I'd not do it automatically. We could either document it or introduce a new parameter.

hofaflo, 2024-03-16:
> For channel 0, the range is approximately 52000µV

With the values below, shouldn't that be 23734 - 26932 = 1802?

```
ch  name    type  unit        min         Q1     median         Q3        max
 0  Fp1     EEG   µV    -28733.96  -28283.90  -28180.43  -27993.25  -26932.31
```

I think the main issue here is that the export uses the same physical range per channel type. This makes sense for data that was initially recorded into an EDF file, with physical min/max set to the actual amplifier limits. However, I'd argue that it's not the best idea for data coming from a format supporting higher resolution (like BDF), as EDF's resolution might not be enough to cover large DC offsets across all signals.

I suggest to allow setting the physical range individually per signal (in the current (unreleased) version of the export code, pass `physical_range=None` [here](https://github.com/mne-tools/mne-python/blob/14ff9483cf92a40479b23f4b17886d3d384f65a5/mne/export/_edf.py#L160)). Additionally, if the current behavior is to be kept as the default, it might make sense to issue a warning in case the resulting resolution is below some threshold deemed "too low".

cbrnr, 2024-03-17:
Ah yes, I missed the minus sign for the maximum 😄. But still, the underlying issue is the signal range being too wide due to offsets in the channels.

A combination of your suggestions might be the best way to proceed. Setting `physical_range=None` can be done in any case, right? Can you let us know when the feature will be available in `edfio`? In addition, if the range in individual channels is still too large for 16 bits, I'd issue a warning (the threshold has to be decided, but I'd say we need at least 0.01µV?).

Teuniz, 2024-03-17:
If you are not willing to get rid of the DC-offset, then forget about converting BDF to EDF.
It's simply not possible without loosing resolution.
If, for whatever reason, you need to keep the DC-offset, then you need to export to BDF instead.
Most of the time, there's a reason why a recording is made with a resolution >16bit and,
in general, it's better to avoid  the conversion from BDF to EDF.

cbrnr, 2024-03-17:
I agree, but technically the offset is irrelevant if the range of a channel can be represented accurately with 16 bits. It's all about the range.

Teuniz, 2024-03-17:
> I agree, but technically the offset is irrelevant if the range of a channel can be represented accurately with 16 bits. It's all about the range.

It cannot, that's why for example Biosemi uses 24 bit adc's. How do you think Biosemi's BDF to EDF converter works?
But feel free to invent the wheel again.

cbrnr, 2024-03-17:
@hofaflo indeed setting `physical_range=None` fixes the issue with my example data. Why did we not do this in the first place and instead set all channels to the global physical min and max?

hofaflo, 2024-03-17:
> Can you let us know when the feature will be available in `edfio`?

Already is, with "current (unreleased) version of the export code" I meant to refer to the code in `mne/export/_edf.py`.

> Why did we not do this in the first place and instead set all channels to the global physical min and max?

This seems to have been initially introduced in 847f6de as a result of the discussion in #9643. I did not change it in #12218 to avoid breaking existing behavior.

cbrnr, 2024-03-18:
Thanks, let's see if the tests still pass if I change it!

cbrnr, 2024-03-18:
@hofaflo before changing the scaling behavior, I would like to make sure that we don't overlook potential issues. [This EEGLAB issue](https://github.com/sccn/eeglab/issues/246) discusses errors that occur when channels are stored with different resolutions (i.e. individual physical min and max) such as not being able to compute a bipolar derivation. However, I think this is due to the subsequent processing steps and not caused by having different resolutions. Furthermore, @Teuniz mentions that the EDF standard mandates that all channels have the same physical min/max, but I did not find this anywhere in the specs.

So I'd really appreciate some insights, and especially maybe a concrete example where channel-specific resolutions can be problematic.

PR #12510 (2024-03-25): Add `physical_range="channelwise"` for EDF export
Fixes #12493 by adding `physical_range="channelwise"`. The default behavior remains `physical_range="auto"`, which sets the physical range per channel *type*. However, in some cases where channels vary wildly in their ranges (mainly due to different offsets), which is often the case if the amplifier used no DC HP filter (e.g. Biosemi), setting individual physical ranges per channel ensures that each channel uses the maximum resolution afforded by the EDF format (only 16 bits). This seems to be perfectly compatible with the EDF standard (please  [... cut]
closed 2024-03-25 by mscheltienne