mne-tools/mne-python #12497: Error in VS Code test discovery

cbrnr, 2024-03-13:
I'm not sure if this is the right place to ask, but test discovery in VS Code suddenly stopped working for me (today) with the following error (Output -> Python):

```
2024-03-13 16:58:14.440 [error] Error discovering pytest tests:
 n [Error]: Traceback (most recent call last):
  File "/Users/clemens/.vscode/extensions/ms-python.python-2024.2.1/pythonFiles/testing_tools/run_adapter.py", line 22, in <module>
    main(tool, cmd, subargs, toolargs)
  File "/Users/clemens/.vscode/extensions/ms-python.python-2024.2.1/pythonFiles/testing_tools/adapter/__main__.py", line 99, in main
    parents, result = run(toolargs, **subargs)
                      ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.vscode/extensions/ms-python.python-2024.2.1/pythonFiles/testing_tools/adapter/pytest/_discovery.py", line 30, in discover
    ec = _pytest_main(pytestargs, [_plugin])
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/_pytest/config/__init__.py", line 174, in main
    ret: Union[ExitCode, int] = config.hook.pytest_cmdline_main(
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/pluggy/_hooks.py", line 501, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/pluggy/_manager.py", line 119, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/pluggy/_callers.py", line 138, in _multicall
    raise exception.with_traceback(exception.__traceback__)
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/pluggy/_callers.py", line 102, in _multicall
    res = hook_impl.function(*args)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/_pytest/main.py", line 332, in pytest_cmdline_main
    return wrap_session(config, _main)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/_pytest/main.py", line 320, in wrap_session
    config.hook.pytest_sessionfinish(
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/pluggy/_hooks.py", line 501, in __call__
    return self._hookexec(self.name, self._hookimpls.copy(), kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/pluggy/_manager.py", line 119, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/pluggy/_callers.py", line 138, in _multicall
    raise exception.with_traceback(exception.__traceback__)
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/pluggy/_callers.py", line 121, in _multicall
    teardown.throw(exception)  # type: ignore[union-attr]
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/_pytest/logging.py", line 871, in pytest_sessionfinish
    return (yield)
            ^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/pluggy/_callers.py", line 121, in _multicall
    teardown.throw(exception)  # type: ignore[union-attr]
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/_pytest/terminal.py", line 866, in pytest_sessionfinish
    result = yield
             ^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/pluggy/_callers.py", line 121, in _multicall
    teardown.throw(exception)  # type: ignore[union-attr]
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/_pytest/warnings.py", line 140, in pytest_sessionfinish
    return (yield)
            ^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/pluggy/_callers.py", line 102, in _multicall
    res = hook_impl.function(*args)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/Projects/mne-python/mne/conftest.py", line 966, in pytest_sessionfinish
    report = item.stash[_phase_report_key]
             ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
  File "/Users/clemens/.pyenv/versions/3.12.2/envs/mne/lib/python3.12/site-packages/_pytest/stash.py", line 80, in __getitem__
    return cast(T, self._storage[key])
                   ~~~~~~~~~~~~~^^^^^
KeyError: <_pytest.stash.StashKey object at 0x31f2dab00>

    at ChildProcess.<anonymous> (/Users/clemens/.vscode/extensions/ms-python.python-2024.2.1/out/client/extension.js:2:226040)
    at Object.onceWrapper (node:events:629:26)
    at ChildProcess.emit (node:events:514:28)
    at maybeClose (node:internal/child_process:1091:16)
    at Process.onexit (node:internal/child_process:302:5)
```

@larsoner any idea what the problem could be?

larsoner, 2024-03-13:
This failing line:
```
 File "/Users/clemens/Projects/mne-python/mne/conftest.py", line 966, in pytest_sessionfinish
    report = item.stash[_phase_report_key]
```
was updated yesterday:

https://github.com/mne-tools/mne-python/blob/2a973338657e6d0609b1d60a8c109cad533c033c/mne/conftest.py#L966-L968

so an update to latest `main` should fix it.

cbrnr, 2024-03-13:
OMG, thanks! May I ask why this change was suddenly necessary?

larsoner, 2024-03-13:
We transitioned away from using `pytest-harvest` to using pytest stashes directly in the last month or so, and this was one of the hiccups in making that change

closed 2024-03-13 by cbrnr