mne-tools/mne-python #12503: Building the MNE doc fails with Python 3.12 and 3.11 (?)

scott-huberty, 2024-03-18:
I installed a fresh development environment recently using the methods suggested in our contributing guide. i.e.:

```
mamba env create --file environment.yml --name mnedev
...
pip install -e ".[dev]"
```

Since we don't specify an upper bound for the python version, mamba installed 3.12. Then running `make html-noplot` resulted in an error fairly early on:

```
  File "/Users/scotterik/miniforge3/envs/mnedev/lib/python3.12/site-packages/latexcodec/codec.py", line 126, in __init__
    self.register_all()
  File "/Users/scotterik/miniforge3/envs/mnedev/lib/python3.12/site-packages/latexcodec/codec.py", line 149, in register_all
    for trans in load_unicode_latex_table():
  File "/Users/scotterik/miniforge3/envs/mnedev/lib/python3.12/site-packages/latexcodec/codec.py", line 102, in load_unicode_latex_table
    with pkg_resources.open_text('latexcodec', 'table.txt') as datafile:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/scotterik/miniforge3/envs/mnedev/lib/python3.12/importlib/resources/_legacy.py", line 18, in wrapper
    warnings.warn(
DeprecationWarning: open_text is deprecated. Use files() instead. Refer to https://importlib-resources.readthedocs.io/en/latest/using.html#migrating-from-legacy for migration advice.
```

<details>
<summary>Full Error Log</summary>

```

# Platform:         darwin; (macOS-14.4-x86_64-i386-64bit)
# Sphinx version:   7.2.6
# Python version:   3.12.2 (CPython)
# Docutils version: 0.20.1
# Jinja2 version:   3.1.3
# Pygments version: 2.17.2

# Last messages:
#   auto_examples/connectivity/index
#   
#   [2K
#   reading sources... [  3%]
#   auto_examples/connectivity/sg_execution_times
#   
#   [2K
#   reading sources... [  3%]
#   auto_examples/datasets/brainstorm_data
#   

# Loaded extensions:
#   sphinx.ext.mathjax (7.2.6)
#   alabaster (0.7.16)
#   sphinxcontrib.applehelp (1.0.8)
#   sphinxcontrib.devhelp (1.0.6)
#   sphinxcontrib.htmlhelp (2.0.5)
#   sphinxcontrib.serializinghtml (1.1.10)
#   sphinxcontrib.qthelp (1.0.7)
#   sphinx.ext.autodoc.preserve_defaults (7.2.6)
#   sphinx.ext.autodoc.type_comment (7.2.6)
#   sphinx.ext.autodoc.typehints (7.2.6)
#   sphinx.ext.autodoc (7.2.6)
#   sphinx.ext.autosummary (7.2.6)
#   sphinx.ext.coverage (7.2.6)
#   sphinx.ext.doctest (7.2.6)
#   sphinx.ext.graphviz (7.2.6)
#   sphinx.ext.intersphinx (7.2.6)
#   sphinx.ext.linkcode (7.2.6)
#   sphinx.ext.todo (7.2.6)
#   matplotlib.sphinxext.plot_directive (3.8.3)
#   numpydoc (1.6.0)
#   sphinx_copybutton (0.5.2)
#   sphinx_design (0.5.0)
[... 153 lines cut ...]
  File "/Users/scotterik/miniforge3/envs/mnedev/lib/python3.12/site-packages/latexcodec/__init__.py", line 1, in <module>
    import latexcodec.codec
  File "/Users/scotterik/miniforge3/envs/mnedev/lib/python3.12/site-packages/latexcodec/codec.py", line 218, in <module>
    _LATEX_UNICODE_TABLE = LatexUnicodeTable(lexer.LatexIncrementalDecoder())
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/scotterik/miniforge3/envs/mnedev/lib/python3.12/site-packages/latexcodec/codec.py", line 126, in __init__
    self.register_all()
  File "/Users/scotterik/miniforge3/envs/mnedev/lib/python3.12/site-packages/latexcodec/codec.py", line 149, in register_all
    for trans in load_unicode_latex_table():
  File "/Users/scotterik/miniforge3/envs/mnedev/lib/python3.12/site-packages/latexcodec/codec.py", line 102, in load_unicode_latex_table
    with pkg_resources.open_text('latexcodec', 'table.txt') as datafile:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/scotterik/miniforge3/envs/mnedev/lib/python3.12/importlib/resources/_legacy.py", line 18, in wrapper
    warnings.warn(
DeprecationWarning: open_text is deprecated. Use files() instead. Refer to https://importlib-resources.readthedocs.io/en/latest/using.html#migrating-from-legacy for migration advice.
```
</details>

I downgraded to python 3.11 and got the same error, then went down to 3.10 (which is what our `build_docs` CI uses) and now building the docs works fine. 

So I think that in order to build the docs with 3.11/12, we'd need to submit a PR to [latexcodec](https://github.com/mcmtroffaes/latexcodec/tree/develop) to use the updated API suggested by [importlib-resources](https://importlib-resources.readthedocs.io/en/latest/using.html#migrating-from-legacy) (?):

https://github.com/mcmtroffaes/latexcodec/blob/2f6a1a82c9e81064a2e9f6cc8739deecd92a4b30/latexcodec/codec.py#L101-L114

This is something I can look into eventually but don't have the time at the moment. I Just wanted to log the issue here in the mean time!

drammock, 2024-03-18:
Is that the full traceback?  It's working for me with python 3.11.7, latexcodec 2.0.1, sphinx 7.2.6, and sphinxcontrib-bibtex 2.6.2 (my best guess at other pkgs that may be relevant). Here's my full sysinfo: 

```console
$ mne sys_info -d
Platform                Linux-5.15.0-58-generic-x86_64-with-glibc2.35
Python                  3.11.7 | packaged by conda-forge | (main, Dec 23 2023, 14:43:09) [GCC 12.3.0]
Executable              /opt/mambaforge/envs/mnedev/bin/python3.11
CPU                     x86_64 (16 cores)
Memory                  125.7 GB

Core
├☑ mne                  1.7.0.dev76+g52a905928 (devel, latest release is 1.6.1)
├☑ numpy                1.26.4 (OpenBLAS 0.3.23.dev with 16 threads)
├☑ scipy                1.11.4
└☑ matplotlib           3.8.2 (backend=QtAgg)

Numerical (optional)
├☑ sklearn              1.3.2
├☑ numba                0.58.1
├☑ nibabel              5.2.0
├☑ nilearn              0.10.2
├☑ dipy                 1.8.0
├☑ openmeeg             2.5.7
├☑ cupy                 13.0.0
└☑ pandas               2.1.4

Visualization (optional)
├☑ pyvista              0.43.2 (OpenGL 4.5.0 NVIDIA 545.23.08 via NVIDIA GeForce RTX 2060/PCIe/SSE2)
├☑ pyvistaqt            0.11.0
├☑ vtk                  9.3.0
├☑ qtpy                 2.4.1 (PyQt6=6.6.0)
├☑ ipympl               0.9.3
├☑ pyqtgraph            0.13.3
├☑ mne-qt-browser       0.7.0.dev9+g547b976
├☑ ipywidgets           8.1.1
├☑ trame_client         2.14.2
├☑ trame_server         2.15.0
├☑ trame_vtk            2.7.0
└☑ trame_vuetify        2.3.1

Ecosystem (optional)
├☑ mne-bids             0.15.0.dev24+g05b8e6b8
├☑ mne-nirs             0.7.0.dev0
├☑ mne-connectivity     0.7.0.dev0
├☑ mne-bids-pipeline    1.6.0.dev8+g9055e9c
├☑ neo                  0.13.0
└☐ unavailable          mne-features, mne-icalabel

Testing
├☑ pytest               8.0.0
├☑ nbclient             0.9.0
├☑ numpydoc             1.7.0rc0.dev0
└☐ unavailable          flake8, pydocstyle

Documentation
├☑ sphinx               7.2.6
├☑ sphinx-gallery       0.16.dev0
└☑ pydata-sphinx-theme  0.15.3dev0

Infrastructure
├☑ decorator            5.1.1
├☑ jinja2               3.1.3
├☑ packaging            23.2
├☑ pooch                1.8.0
└☑ tqdm                 4.66.1
```

Here's

larsoner, 2024-03-18:
... and really this is a case where we should probably also add the `DeprecationWarning` to the list of ignores in `doc/conf.py`. By default we turn all warnings into errors during our doc build, but we shouldn't need to worry so much about this one (it's not in our code), though of course we should look into upstream fixes!

scott-huberty, 2024-03-18:
> Is that the full traceback? It's working for me with python 3.11.7, latexcodec 2.0.1, sphinx 7.2.6, and sphinxcontrib-bibtex 2.6.2 (my best guess at other pkgs that may be relevant). Here's my full sysinfo:

Weird!  _Maybe_ this issue was just a fluke when I set up my new env, since the docs build with python 3.11 for you. Feel free to close this issue.

And FYI the full traceback of my error is in the "Full error log" dropdown in my initial message.

<br>

FWIW here is my package info (**TL;DR: the package versions look pretty much the same**):

| 3.11.8 (doesn't work)  | 3.10.0 (works) |
| ------------- | ------------- |
| latexcodec 3.0.0  |  latexcodec 3.0.0  |
| Sphinx 7.2.6  | Sphinx 7.2.6 |
| sphinxcontrib-bibtex  2.6.2 |  sphinxcontrib-bibtex  2.6.2  |
| importlib_resources 6.3.0 | importlib_resources 6.3.0 |

<details>
<summary>sys info for python 3.11</summary>

```
Platform             macOS-14.4-x86_64-i386-64bit
Python               3.11.8 | packaged by conda-forge | (main, Feb 16 2024, 20:51:20) [Clang 16.0.6 ]
Executable           /Users/scotterik/miniforge3/envs/mnedev/bin/python3.11
CPU                  i386 (8 cores)
Memory               32.0 GB

Core
├☑ mne               1.7.0.dev149+g2a9733386.d20240318 (devel, latest release is 1.6.1)
├☑ numpy             1.26.4 (OpenBLAS 0.3.26 with 4 threads)
├☑ scipy             1.12.0
└☑ matplotlib        3.8.3 (backend=MacOSX)

Numerical (optional)
├☑ sklearn           1.4.1.post1
├☑ numba             0.59.0
├☑ nibabel           5.2.1
├☑ nilearn           0.10.3
├☑ dipy              1.9.0
├☑ openmeeg          2.5.7
├☑ pandas            2.2.1
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.43.4 (OpenGL 4.1 INTEL-22.5.10 via Intel(R) Iris(TM) Plus Graphics OpenGL Engine)
├☑ pyvistaqt         0.11.0
├☑ vtk               9.2.6
├☑ qtpy              2.4.1 (PyQt5=5.15.8)
├☑ ipympl            0.9.3
├☑ pyqtgraph         0.13.4
├☑ mne-qt-browser    0.6.1
├☑ ipywidgets        8.1.2
├☑ trame_client      2.16.3
├☑ trame_server      2.17.2
├☑ trame_vtk         2.8.5
└☑ trame_vuetify     2.4.3

Ecosystem (optional)
├☑ mne-bids          0.14
├☑ mne-connectivity  0.6.0
├☑ neo               0.13.0
└☐ unavailable       mne-nirs, mne-features, mne-icalabel, mne-bids-pipeline
```
</details>

<details>
<summary>sys info for python 3.10</summary>

```
Platform             macOS-14.4-x86_64-i386-64bit
Python               3.10.0 | packaged by conda-forge | (default, Nov 20 2021, 02:43:39) [Clang 11.1.0 ]
Executable           /Users/scotterik/miniforge3/envs/mnedev_310/bin/python3.10
CPU                  i386 (8 cores)
Memory               32.0 GB

Core
├☑ mne               1.7.0.dev149+g2a9733386.d20240318 (devel, latest release is 1.6.1)
├☑ numpy             1.26.4 (OpenBLAS 0.3.26 with 4 threads)
├☑ scipy             1.12.0
└☑ matplotlib        3.8.3 (backend=MacOSX)

Numerical (optional)
├☑ sklearn           1.4.1.post1
├☑ numba             0.59.0
├☑ nibabel           5.2.1
├☑ nilearn           0.10.3
├☑ dipy              1.9.0
├☑ openmeeg          2.5.7
├☑ pandas            2.2.1
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.43.4 (OpenGL 4.1 INTEL-22.5.10 via Intel(R) Iris(TM) Plus Graphics OpenGL Engine)
├☑ pyvistaqt         0.11.0
├☑ vtk               9.2.6
├☑ qtpy              2.4.1 (PyQt5=5.15.8)
├☑ ipympl            0.9.3
├☑ pyqtgraph         0.13.4
├☑ mne-qt-browser    0.6.1
├☑ ipywidgets        8.1.2
├☑ trame_client      2.16.3
├☑ trame_server      2.17.2
├☑ trame_vtk         2.8.5
└☑ trame_vuetify     2.4.3

Ecosystem (optional)
├☑ mne-bids          0.14
├☑ mne-connectivity  0.6.0
├☑ neo               0.13.0
└☐ unavailable       mne-nirs, mne-features, mne-icalabel, mne-bids-pipeline
```
</details>

If I have some time I can try to investigate further, or see if I replicate my issue when creating another fresh environment with 3.12.

drammock, 2024-03-18:
Sorry for missing the collapsed traceback before. The other important moving piece is `pybtex`. See e.g.:

- https://bitbucket.org/pybtex-devs/pybtex/issues/169/replace-pkg_resources-with
- https://github.com/mcmtroffaes/sphinxcontrib-bibtex/issues/345#issuecomment-1900769572

TL;DR: pybtex is unmaintained, sphinxcontrib-bibtex hasn't yet sorted out what to do about it, and so (I think) if you weren't getting this error from `latexcodec` you would likely get a very similar error (on Py3.12) from pybtex.  I'm still not sure why it failed for you on 3.11 though.

scott-huberty, 2024-03-18:
> Sorry for missing the collapsed traceback before. The other important moving piece is `pybtex`. See e.g.:
> 
> * https://bitbucket.org/pybtex-devs/pybtex/issues/169/replace-pkg_resources-with
> * [Cannot import due to `pkg_resources` on Python 3.12 mcmtroffaes/sphinxcontrib-bibtex#345 (comment)](https://github.com/mcmtroffaes/sphinxcontrib-bibtex/issues/345#issuecomment-1900769572)
> 
> TL;DR: pybtex is unmaintained, sphinxcontrib-bibtex hasn't yet sorted out what to do about it, and so (I think) if you weren't getting this error from `latexcodec` you would likely get a very similar error (on Py3.12) from pybtex. I'm still not sure why it failed for you on 3.11 though.

Awh, ok, thanks @drammock ! 

> I'm still not sure why it failed for you on 3.11 though.

Me neither, but I created the environment with 3.12 and then downgraded to 3.11, so maybe there were some package versions that were lingering from python 3.12 that didn't change when downgrading (i.e. if I create a fresh 3.11 environment maybe I won't get this issue 🤷 ).

Anyways, I'll close this issue. @larsoner if you do think it's worth adding the `DeprecationWarning` to `conf.py` let me know and I can open a PR.

closed 2024-03-18 by scott-huberty