mne-tools/mne-python #12517: GDF reader bug

anamelnich, 2024-03-26:
The lowpass filter in raw.info is not set correctly when uploading raw gdf file and doesn't update after filtering. I noticed it while trying to plot continuous data and getting a ZeroDivision error.

Code to replicate:
```
import mne

file_path = '2024-02-28_13_14_s01_s01_r01.gdf'
raw = mne.io.read_raw_gdf(file_path, preload=True)
print(raw.info)
#raw.plot()

raw.filter(1, 50, fir_design="firwin")
print(raw.info)
raw.plot()

```
Error:
```
Traceback (most recent call last):
  File "...eegplot_test.py", line 11, in <module>
    raw.plot()
  File "...MNE-Python/1.6.1_0/.mne-python/lib/python3.11/site-packages/mne/io/base.py", line 1808, in plot
    return plot_raw(
           ^^^^^^^^^
  File "<decorator-gen-157>", line 12, in plot_raw
  File "...MNE-Python/1.6.1_0/.mne-python/lib/python3.11/site-packages/mne/viz/raw.py", line 258, in plot_raw
    decim, picks_data = _handle_decim(info, decim, lowpass)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "...MNE-Python/1.6.1_0/.mne-python/lib/python3.11/site-packages/mne/viz/utils.py", line 1987, in _handle_decim
    decim = max(int(info["sfreq"] / (lp * 3) + 1e-6), 1)
                    ~~~~~~~~~~~~~~^~~~~~~~~~
ZeroDivisionError: float division by zero
```
raw.info before filtering:
```
<Info | 8 non-empty values
 bads: []
 ch_names: FP1, FPZ, FP2, F7, F3, FZ, F4, F8, FC5, FC1, FC2, FC6, M1, T7, ...
 chs: 67 EEG, 1 Stimulus
 custom_ref_applied: False
 highpass: 1.0 Hz
 lowpass: 0.0 Hz
 meas_date: 2024-02-28 19:14:52 UTC
 nchan: 68
 projs: []
 sfreq: 512.0 Hz
 subject_info: 4 items (dict)
>
```
raw.info after filtering:
```
<Info | 8 non-empty values
 bads: []
 ch_names: FP1, FPZ, FP2, F7, F3, FZ, F4, F8, FC5, FC1, FC2, FC6, M1, T7, ...
 chs: 67 EEG, 1 Stimulus
 custom_ref_applied: False
 highpass: 1.0 Hz
 lowpass: 0.0 Hz
 meas_date: 2024-02-28 19:14:52 UTC
 nchan: 68
 projs: []
 sfreq: 512.0 Hz
 subject_info: 4 items (dict)
>
```
Other information: 
MNE version: 1.6.1
Operating system: macOS 12

cbrnr, 2024-03-26:
Note that I get the following warnings when loading the file:

```
RuntimeWarning: Channels contain different highpass filters. Highest filter setting will be stored.
RuntimeWarning: Channels contain different lowpass filters. Lowest filter setting will be stored.
```

larsoner, 2024-03-27:
We could either restrict `info["lowpass"]` to be strictly less than zero or (my vote) improve the sufficient-lowpass check to only run if `info["lowpass"] > 0`

cbrnr, 2024-03-27:
Whatever you end up deciding, I just confirmed with @schloegl that the file in question does indeed have all zeros for its filter settings, so we need to deal with such files gracefully.

larsoner, 2024-03-27:
Yeah if we know this can happen with GDF files on some systems we could detect the behavior for at least that format

cbrnr, 2024-03-27:
I think even with EDF it is not guaranteed that those header fields contain sane values. We could use fallback values of 0 Hz and fs/2 Hz in such cases, respectively.