mne-tools/mne-python #12521: BUG: EDF export failing on latest edfio

larsoner, 2024-03-29:
Our use of private `edfio._utils` is problematic, @cbrnr see [failed CI run](https://dev.azure.com/mne-tools/mne-python/_build/results?buildId=29598&view=logs&jobId=dded70eb-633c-5c42-e995-a7f8d1f99d91&j=dded70eb-633c-5c42-e995-a7f8d1f99d91&t=1ccededa-7e7e-5a92-87ce-e531d760d025) :
```
mne\export\tests\test_export.py:305: in test_channel_label_too_long_for_edf_raises_error
    raw.export(tmp_path / "test.edf")
<decorator-gen-346>:12: in export
    ???
mne\io\base.py:1792: in export
    export_raw(
<decorator-gen-599>:12: in export_raw
    ???
mne\export\_export.py:74: in export_raw
    from ._edf import _export_raw
mne\export\_edf.py:14: in <module>
    from edfio._utils import round_float_to_8_characters  # noqa: E402
E   ModuleNotFoundError: No module named 'edfio._utils'
```

larsoner, 2024-03-29:
Either way is fine with me but a fix that would work for all edfio versions would be to copy the function over here (especially if it's only a couple of lines). I was about to add a skip in #12520 but instead I'll try that!

hofaflo, 2024-03-29:
:+1:
It's rather short and can be simplified a bit as mne only uses one way of rounding:
```python
def _round_float_to_8_characters(value):
    if isinstance(value, int) or value.is_integer():
        return value
    length = 8
    integer_part_length = str(value).find(".")
    if integer_part_length == length:
        return round(value)
    factor = 10 ** (length - 1 - integer_part_length)
    return round(value * factor) / factor
```

PR #12520 (2024-03-29): MAINT: Restore OMP_NUM_THREADS=2 on Windows
In theory this should work now that https://github.com/scikit-learn/scikit-learn/pull/28692 has landed and new wheels presumably uploaded

Closes #12521
closed 2024-03-29 by larsoner