mne-tools/mne-python #12530: set_annotations changes key:value mapping in event_id dict

OleBialas, 2024-04-05:
### Description of the problem

I created new annotations using `annotations_from_events` and wrote them to the raw data using `set_annotations`. It worked as expected, with the exception that the mapping between keys and values in the event description dictionary got scrambled.

### Steps to reproduce

```Python
from mne import events_from_annotations, annotations_from_events
from mne.io import read_raw_edf
from mne.datasets import eegbci

# load data and extract events
path = eegbci.load_data(subject=1, runs=[3])[0]
raw = read_raw_edf(path)
events, event_id = events_from_annotations(raw)
print(event_id, "\n", events)

# create annotations with new event description
event_desc = {1: "one", 2: "two", 3: "three"}
annotations = annotations_from_events(events, raw.info['sfreq'], event_desc)
print(annotations.to_data_frame())  # annotations are correct

# set annotations, then extract and print events
raw.set_annotations(annotations)
events, event_id = events_from_annotations(raw)
print(event_id, "\n", events) # now the keys and values got mixed up
```

### Expected results

After setting the annotations, the event IDs returned by `events_from_annotations` should be the same as the `event_desc` passed to `annotations_from_events`

### Actual results

The key:value mapping in the event IDs is different from the mapping specified in the `event_desc`

### Additional information

Platform             Linux-6.8.2-arch2-1-x86_64-with-glibc2.39
Python               3.9.16 | packaged by conda-forge | (main, Feb  1 2023, 21:39:03)  [GCC 11.3.0]
Executable           /home/obi/miniforge-pypy3/envs/eeg/bin/python3.9
CPU                   (8 cores)
Memory               29.3 GB

Core
├☑ mne               1.4.2
├☑ numpy             1.24.4 (OpenBLAS 0.3.21 with 8 threads)
├☑ scipy             1.11.1
├☑ matplotlib        3.7.2Installed tk event loop hook.
 (backend=TkAgg)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.3.0
├☑ numba             0.57.1
├☑ pandas            2.0.3
└☐ unavailable       nibabel, nilearn, dipy, openmeeg, cupy

Visualization (optional)
└☐ unavailable       pyvista, pyvistaqt, ipyvtklink, vtk, qtpy, ipympl, pyqtgraph, mne-qt-browser

Ecosystem (optional)
├☑ mne-bids          0.13
└☐ unavailable       mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

larsoner, 2024-04-05:
Does it work okay if in your last call you add `event_id=event_id` like:
```
events, event_id = events_from_annotations(raw, event_id=event_id)
```
? In principle the mapping from annotations to event id numbers is somewhat arbitrary, and if you don't pass `event_id` then they might just get assigned in the order they appear in the annotations.

But without the file it's tough to know if this is the problem that you're actually hitting -- can you share the file publicly (or if not, privately with me)?

OleBialas, 2024-04-05:
No, adding the event_id does not change the result. Why would the mapping be arbitrary if, in the `event_desc` argument, you explicitly define a mapping, is that just ignored? I used public data for this example, you can download it with `mne.datasets.eegbci.load_data()`

larsoner, 2024-04-05:
Sorry my suggestion was off, you need something more like this:
```
event_id = {v: k for k, v in event_desc.items()}
events, event_id = events_from_annotations(raw, event_id=event_id)
```
then things are okay I think, your `print` at the end then produces:
```
{np.str_('one'): 1, np.str_('three'): 3, np.str_('two'): 2} 
 [[    0     0     1]
 [  672     0     3]
 [ 1328     0     1]
 [ 2000     0     2]
```
which matches the dataframe one
```
                     onset  duration description
0  1970-01-01 00:00:00.000       0.0         one
1  1970-01-01 00:00:04.200       0.0       three
2  1970-01-01 00:00:08.300       0.0         one
3  1970-01-01 00:00:12.500       0.0         two
```
So to explain a bit more:

> Why would the mapping be arbitrary if, in the event_desc argument, you explicitly define a mapping, is that just ignored?

In your original code, `event_desc` is used to map event ID integers to annotation names and create `annotations`. At that point, `annotations` has no knowledge anymore about what numerical event ID integers were used to create them, for example the first one:
```
>>> annotations[0]
OrderedDict([('onset', np.float64(0.0)), ('duration', np.float64(0.0)), ('description', np.str_('one')), ('orig_time', None)])
```
It just has an onset, duration, description, and orig_time. Nothing about the numerical event ID. So then one line later in your code when you do:
```
events, event_id = events_from_annotations(raw)
```
MNE will look at `raw.annotations` and start assigning integers to annotation descriptions. It isn't exactly *arbitrary* like I said -- it will happen in the order the annotations descriptions show up. So the first event is "one", so this gets assigned the numerical event ID 1. The next is "three", so this gets assigned 2. If you don't want MNE to assign numerical IDs based on the order they show up in `raw.annotations`, then you need to give it an explicit `event_id` arg to tell it the mapping from `annotation.description`s to those numbers. Hence why the two lines at the top of this comment fix things. Make sense?

larsoner, 2024-04-05:
> It isn't exactly arbitrary like I said -- it will happen in the order the annotations descriptions show up

Argh this part is wrong, too! I need to read our docs more closely. The default behavior is `"auto"` which for EDF files is the same as `None`, namely:
```
        - **None**: Map descriptions to unique integer values based on their
          ``sorted`` order.
```
The original annotations have descriptions in `sorted` order of `['T0', 'T1', 'T2']`. So in the first `events_from_annotations` these get `event_id = {'T0': 1, 'T1': 2, 'T2': 3}`. Then you reassign the annotations with one/two/three, which in `sorted` order are `['one', 'three', 'two']`, so the `event_id="auto"` mapping does something you don't expect.

In any case, I think the solution here is to use an explicit `event_id` in calls to `events_from_annotations` -- ideally in both calls actually! -- to have things make the most sense.

OleBialas, 2024-04-07:
Thanks a lot for your detailed explanation of how the annotations work!

closed 2024-04-07 by OleBialas