mne-tools/mne-python #12542: Be more explicit about what is returned by read_epochs_eeglab and read_epochs_kit

nabilalibou, 2024-04-16:
### Proposed documentation enhancement

It's a minor detail, but the docstrings for functions like [mne.read_epochs_eeglab()](https://mne.tools/stable/generated/mne.read_epochs_eeglab.html) and [mne.read_epochs_kit()](https://mne.tools/stable/generated/mne.read_epochs_kit.html) indicate that an [Epochs instance](https://mne.tools/stable/generated/mne.Epochs.html#mne.Epochs) is returned, while in reality they return subclasses of `BaseEpochs`; specifically `EpochsEEGLAB `and `EpochsKIT`, respectively. 
It can be deceptive, for example it could encourage the use of a ` isinstance(epochs, mne.Epoch)` because although `BaseEpoch` exists, the documentation could misleads into thinking that there is only one type of `Epoch` class.

Even though there don't seem to be any html pages created for these subclasses that could be linked, i think we could add a few details to be clearer, what do you think ?

hoechenberger, 2024-04-16:
this problem affects many readers and I believe it has at least been brought up when we added some type hints.

i agree with you that this should be improved.

larsoner, 2024-04-16:
For now the easiest solution would be to say that the return type is BaseEpochs, which is nice because it [is a documented type](https://mne.tools/stable/generated/mne.BaseEpochs.html)

nabilalibou, 2024-04-16:
A solution in the meantime to make sure the reader has all the keys needed to understand what's behind could be something like:

```
Returns:
   EpochsEEGLAB/EpochsKIT : instance of [BaseEpochs]
        EpochsEEGLAB/EpochsKIT object (subclass of [BaseEpochs])
```
And having a markdown box: 
```
See also: 
   [mne.Epochs] Documentation of attributes and methods.
```

closed 2024-06-04 by nabilalibou