mne-tools/mne-python #12543: Unicode support in the method add.text() of mne.viz.brain

Jieli12, 2024-04-16:
### Describe the new feature or enhancement

When I run the following command:
`brain.add_text(0.1, 0.9, "chinese\u2190中文", color="black", font_size=10)`

The added text only shows `chinese`, see below
[image]

However, I expect the added text to be `chinese <- 中文`. Would you please add this feature?

### Describe your proposed implementation

I am not a Python expert, but I found that the Python package `PyVista` also has the method `add_text()` [here](https://docs.pyvista.org/version/stable/api/plotting/_autosummary/pyvista.Plotter.add_text.html). Maybe it can offer some thought on how to add this new feature.

### Describe possible alternatives

None.

drammock, 2024-05-10:
Sorry for the slow reply!  This is a bug of PyVista, not MNE-Python.  Under the hood we are already calling exactly the `pyvista.Plotter.add_text()` method that you linked to.  Note that if you modify [this PyVista tutorial](https://tutorial.pyvista.org/tutorial/03_figures/solutions/b_lighting_mesh.html) to use your text, it also fails to print the arrow and the hanzi:

```py
import pyvista as pv
from pyvista import examples

mesh = examples.download_st_helens().warp_by_scalar()
cpos = [(575848.0, 5128459.0, 22289.0), (562835.0, 5114981.5, 2294.5), (-0.5, -0.5, 0.7)]
p = pv.Plotter(shape=(1, 2), window_size=[1500, 500])

p.subplot(0, 0)
p.add_mesh(mesh, show_scalar_bar=False)
p.add_text("chinese\u2190中文")  # <- CHANGED HERE

p.subplot(0, 1)
specular = 1.0
p.add_mesh(mesh, specular=specular, show_scalar_bar=False)
p.add_text(f'Specular of {specular}')

p.link_views()
p.view_isometric()
p.show(cpos=cpos)
```

[image]

I recommend opening a similar issue on PyVista repository.

Jieli12, 2024-05-10:
Would you please run the following code?
```
import pyvista as pv
from pyvista import examples

font = "Microsoft YaHei"
mesh = examples.download_st_helens().warp_by_scalar()
cpos = [
    (575848.0, 5128459.0, 22289.0),
    (562835.0, 5114981.5, 2294.5),
    (-0.5, -0.5, 0.7),
]
p = pv.Plotter(shape=(1, 2), window_size=[1500, 500])

p.subplot(0, 0)
p.add_mesh(mesh, show_scalar_bar=False)
p.add_text(
    "chinese\u2190中文", font_file="/System/Library/Fonts/STHeiti Medium.ttc"
)  # <- CHANGED HERE, specifying the  `font_file`

p.subplot(0, 1)
specular = 1.0
p.add_mesh(mesh, specular=specular, show_scalar_bar=False)
p.add_text(f"Specular of {specular}")

p.link_views()
p.view_isometric()
p.show(cpos=cpos)
```
It shows
[image]

Could you please also add the argument `font_file` in [`brain.add_text()` function](https://mne.tools/stable/generated/mne.viz.Brain.html#mne.viz.Brain.add_text)?