mne-tools/mne-python #12547: Bad channels do not show up at all in `raw.plot()`

cbrnr, 2024-04-17:
I don't know if this is an intentional change, but bad channels used to be drawn in gray, and now they don't show up at all:

```python
from numpy.random import default_rng

from mne import create_info
from mne.io import RawArray

def create_toy_data(n_channels=3, duration=25, sfreq=250, seed=None):
    rng = default_rng(seed)
    data = rng.standard_normal(size=(n_channels, duration * sfreq)) * 5e-6
    info = create_info(n_channels, sfreq, "eeg")
    return RawArray(data, info)

raw = create_toy_data()
raw.info["bads"] = ["1"]
raw.plot()  # channel 2 (labeled "1") is missing!
```

CC @hoechenberger

hoechenberger, 2024-04-17:
Thanks for reporting this, @cbrnr. I ran into this today and I thought I was going crazy, constantly being unable to find certain channels – until I discovered they were bad and wouldn't even show up with `picks="all"`, `picks="eeg"`. Had to set `info["bass"] = []` to make them appear again.

larsoner, 2024-04-17:
Can you see if changing this line

https://github.com/mne-tools/mne-python/pull/12467/files#diff-463b263c18142ec9504e9eef4ac62f3882d14cf0f3b6dfd001b0dc00d582ada7R316

To have `exclude=()` fixes the issue? I'm guessing it will

cbrnr, 2024-04-17:
Yes it does, thanks @larsoner. We should double-check that other plot methods are not affected such as `epochs.plot()` though.

closed 2024-04-17 by larsoner