mne-tools/mne-python #12551: Mouse Offset in Brain Plot

gedwald, 2024-04-18:
### Description of the problem

I raised this subject two days ago in Discourse group:
https://mne.discourse.group/t/problem-with-stc-plot-why-does-the-mouse-have-a-variable-offset-on-the-brain-image-plot/8648

I am new to NME and Dipoles, so I hesitate to post this here, fearing I may be overlooking some simple transform error.

The problem is that my mouseclicks on the brain plot create dipoles in locations offset from the mouse, and the offset is not constant.  Attached ExhibitB.png and ExhibitC.png show the offset from clicking at two locations and the appearance of the Orange and Green bubbles.  In general, the further to the right I clicked, the larger the horizontal offset.  The vertical offset remained constant, I think. 

[image]
[image]
[DisplayOffsetDemo.txt](https://github.com/mne-tools/mne-python/files/15012518/DisplayOffsetDemo.txt)

### Steps to reproduce

```Python
...
fwd = mne.make_forward_solution( evoked.info, trans=None, src=src, bem=bem, verbose=True )
inv = mne.minimum_norm.make_inverse_operator(evoked.info, fwd, cov, verbose=True)
stc = mne.minimum_norm.apply_inverse(evoked, inv)
brain = stc.plot(subjects_dir=subjects_dir, initial_time=0.0, hemi='lh', views="medial")
...

The python contents of the minimal Jupyter notebook is attached (DisplayOffsetDemo.txt).
The notebook kcannot be attached - what format is convenient for you?  The system response 
could be useful to you, but is obviously missing from this listing.
```

### Link to data

Somewhat cumbersome to reduce the data, the full size is around 1.3GB.
If deemed helpful, I'll find a way to reduce the data and repeat the process with a smaller set of epochs

### Expected results

Dipole 'bubble' should be created where I clicked the mouse on the brain plot.

### Actual results

Dipole bubble appears at considerable offsets away from the click location.

### Additional information

I am using Ubuntu 22.04, and MNE 1.6.1.
I believe I am not using any GPU.  The machine has an Nvidia 1660 card, but it is not used for the diplays, nor is it configured to be used in any way I can tell. 

However, the setup on my system must be wrong, as it produces graphics artefacts that cannot be common.
(see Exhibits "A" and "D" attached above)
[image]
[image]

Further, a similar effect is seen by running the same notebook on a macOS M1 machine, with OS 14.4.1.
That Mac instance is using MNE 1.6.1 as well, and "pysvistaqt" as backend.
I believe the Ubuntu machine has pyvistaqt, but I must admit I am not entirely sure, and don't know how to find out.  
See last attachment "OffsetMacOS_pyvistaqt3d".

[image]

larsoner, 2024-04-18:
What is your `mne sys_info`? It's possible this is a bug with PyQt5 and HiDPI (though it's more common on macOS than Linux)

gedwald, 2024-04-18:
Thank you for the query; these are my two environments:

::: MacOS :::

Platform             macOS-14.4.1-arm64-arm-64bit
Python               3.9.6 (default, Feb  3 2024, 15:58:27)  [Clang 15.0.0 (clang-1500.3.9.4)]
Executable           /Users/edwald/python_venvs/mne/bin/python3
CPU                  arm (8 cores)
Memory               16.0 GB

Core
├☑ mne               1.6.1 (latest release)
├☑ numpy             1.26.4 (OpenBLAS 0.3.23.dev with 8 threads)
├☑ scipy             1.12.0
├☑ matplotlib        3.8.3 (backend=module://matplotlib_inline.backend_inline)
├☑ pooch             1.8.0
└☑ jinja2            3.1.3

Numerical (optional)
├☑ sklearn           1.4.1.post1
├☑ nibabel           5.2.0
├☑ nilearn           0.10.3
├☑ pandas            2.2.0
└☐ unavailable       numba, dipy, openmeeg, cupy

Visualization (optional)
├☑ pyvista           0.43.3 (OpenGL 4.1 Metal - 88 via Apple M1)
├☑ pyvistaqt         0.11.0
├☑ vtk               9.3.0
├☑ qtpy              2.4.1 (PyQt6=6.6.1)
├☑ pyqtgraph         0.13.4
├☑ mne-qt-browser    0.6.2
├☑ ipywidgets        8.1.2
└☐ unavailable       ipympl, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

::: Ubuntu 22.04  :::
>>> mne.sys_info()
Platform             Linux-6.5.0-25-generic-x86_64-with-glibc2.35
Python               3.11.0rc1 (main, Aug 12 2022, 10:02:14) [GCC 11.2.0]
Executable           /home/edwaldt20ni/VirtualEnvs/neweeg/bin/python
CPU                  x86_64 (12 cores)
Memory               62.6 GB

Core
├☑ mne               1.6.1 (latest release)
├☑ numpy             1.26.4 (OpenBLAS 0.3.23.dev with 12 threads)
├☑ scipy             1.12.0
├☑ matplotlib        3.8.3 (backend=QtAgg)
├☑ pooch             1.8.1
└☑ jinja2            3.1.3

Numerical (optional)
├☑ sklearn           1.4.1.post1
├☑ nibabel           5.2.0
├☑ nilearn           0.10.3
├☑ pandas            2.2.0
└☐ unavailable       numba, dipy, openmeeg, cupy

Visualization (optional)
├☑ pyvista           0.43.3 (OpenGL 4.5 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2 via llvmpipe (LLVM 15.0.7, 256 bits))
├☑ pyvistaqt         0.11.0
├☑ vtk               9.2.6
├☑ qtpy              2.4.1 (PyQt6=6.6.1)
├☑ pyqtgraph         0.13.3
├☑ mne-qt-browser    0.6.1
├☑ ipywidgets        8.1.2
└☐ unavailable       ipympl, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

larsoner, 2024-04-19:
> If deemed helpful, I'll find a way to reduce the data and repeat the process with a smaller set of epochs

Can you try with this snippet (will download ~1.5GB):
```
>>> import mne
>>> data_path = mne.datasets.sample.data_path(verbose=True)
>>> stc = mne.read_source_estimate(data_path / "MEG" / "sample" / "sample_audvis-meg-eeg-lh.stc", subject="sample")
>>> stc.plot()
```
For me it produces something that interacts as expected, does it work for you?

[image]

gedwald, 2024-04-19:
Not quite:
I haven’t had opportunity to test this on the Ubuntu box, but here’s what happens on the Mac.
I don’t know why it’s looking for an -fsaverage- file 

(mne) edwald@temmi ~ % python 
Python 3.9.6 (default, Feb  3 2024, 15:58:27) 
[Clang 15.0.0 (clang-1500.3.9.4)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import mne
>>> data_path=mne.datasets.sample.data_path(verbose=True)
Downloading file 'MNE-sample-data-processed.tar.gz' from 'https://osf.io/86qa2/download?version=6' to '/Users/edwald/mne_data'.
/Users/edwald/python_venvs/mne/lib/python3.9/site-packages/urllib3/__init__.py:35: NotOpenSSLWarning: urllib3 v2 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(
100%|█████████████████████████████████████| 1.65G/1.65G [00:00<00:00, 23.3TB/s]
Untarring contents of '/Users/edwald/mne_data/MNE-sample-data-processed.tar.gz' to '/Users/edwald/mne_data'
Download complete in 01m19s (1576.2 MB)
>>> stc=mne.read_source_estimate(data_path / "MEG" / "sample" / "sample_audvis-meg-eeg-lh.stc", subject="sample")
>>> stc.plot()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/edwald/python_venvs/mne/lib/python3.9/site-packages/mne/source_estimate.py", line 743, in plot
    brain = plot_source_estimates(
  File "<decorator-gen-165>", line 12, in plot_source_estimates
  File "/Users/edwald/python_venvs/mne/lib/python3.9/site-packages/mne/viz/_3d.py", line 2398, in plot_source_estimates
    subjects_dir = get_subjects_dir(subjects_dir=subjects_dir, raise_error=True)
  File "/Users/edwald/python_venvs/mne/lib/python3.9/site-packages/mne/utils/config.py", line 475, in get_subjects_dir
    subjects_dir = _check_fname(
  File "<decorator-gen-0>", line 12, in _check_fname
  File "/Users/edwald/python_venvs/mne/lib/python3.9/site-packages/mne/utils/check.py", line 263, in _check_fname
    raise FileNotFoundError(f'{name} does not exist: "{fname}"')
FileNotFoundError: subjects_dir does not exist: "/Users/edwald/mne_data/MNE-fsaverage-data"
>>> data_path
PosixPath('/Users/edwald/mne_data/MNE-sample-data')
>>>

larsoner, 2024-04-19:
Argh sorry that last line in the MWE should be:
```
>>> stc.plot(subjects_dir=data_path / "subjects")
```

gedwald, 2024-04-19:
Thank you - 
the plot appears, but I can only get new dipole bubbles on the brain if I click the mouse on the extreme right edge 
Ie. the V1 area and further up the edge.  Then I get the bubbles on the lower edge of the brain.  Any clicks further inwards on the brain do not visibly produce anything, probably because the bubble falls off the left edge of the brain plot. ( I have not rotated the plot at all).

larsoner, 2024-04-22:
Okay I was able to replicate on my macOS machine -- can you see if being on this branch of PyVistaQt fixes things for you on mac at least?

https://github.com/pyvista/pyvistaqt/pull/565

And maybe we'll get lucky and it'll also fix things on Linux.

gedwald, 2024-04-22:
I’d happily do that, but I don’tknow how. Just clone a repository and build?best regardsT. Edwald

pyvista/pyvistaqt#565

larsoner, 2024-04-22:
You should be able to do `pip install git+https://github.com/larsoner/pyvistaqt.git@pyqt6` to install my branch

gedwald, 2024-04-22:
This WORKS!  Or, at least- the dipole bubble now appears exactly where I click with the mouse.  
Thank you very much for this wizardry.  Brilliant!
Tomorrow I will have the opportunity to test this on Ubuntu as well, or anything else I can contribute.

gedwald, 2024-04-23:
The code works correctly on Ubuntu as well, with your fix. 
Both my own code and your example above. 
Many thanks.

closed 2024-04-23 by larsoner