mne-tools/mne-python #12553: Incorrect rendering of embedded HTML (?)

hoechenberger, 2024-04-19:
### Proposed documentation enhancement

See e.g. https://mne.tools/dev/auto_tutorials/intro/70_report.html

[image]

drammock, 2024-04-19:
bug is there in stable docs too 😢

hoechenberger, 2024-04-28:
Changing the `flex-direction` of `p.rubric` from `column` to `row` fixes the problem for me:

[image]

hoechenberger, 2024-04-28:
When I simply drop the `p.rubric` styling altogether it looks as intended:

[image]

hoechenberger, 2024-04-28:
That may need fixing in pydata-sphinx-theme

PR #12583 (2024-05-21): MRG: Revamp HTML reprs of `Raw`, `Epochs`, `Evoked`, and `Info`
For example, the `Evoked` repr contained the number of channels, but failed to subtract or report any bads.

Also I believe reporting the number of sampling points is quite pointless , so I removed that as well.

The table layout wasn't up-to-date either.

`main` (striped layout only added by VS Code, actually not present in the tables we produce!)

[image]

This PR:

[image]

For comparison, `Epochs`:

[image]

MWE:
```python
# %%
import mne

sample_dir = mne.datasets.sample.data_path()
sample_fname = sample_dir / "MEG" / "sample [... cut]
closed 2024-05-21 by drammock