mne-tools/mne-python #12559: Add SOUND and SSP-SIR algorithms

matteo-d-m, 2024-04-22:
Dear MNE-Python developers, 

I am Matteo De Matola, doctoral student at the University of Trento (Italy).

I have noticed that MNE-Python does not include an implementation of the SOUND ([Mutanen et al., 2018](https://www.sciencedirect.com/science/article/pii/S1053811917308480)) and SSP-SIR ([Mutanen et al., 2016](https://www.sciencedirect.com/science/article/pii/S1053811916301495)) algorithms. These algorithms are widely used in TMS-EEG preprocessing, but to the best of my knowledge, they are currently implemented in MATLAB only.

The absence of a reliable Python implementation for SOUND and SSP-SIR does not facilitate the adoption of MNE-Python in the TMS-EEG community. In agreement with Tuomas Mutanen, the inventor of the two algorithms, I volunteer to translate them from MATLAB to Python and integrate them into MNE. 

---

I did not open an ENH-type issue because:

1. I do not have a proposed implementation yet
2. I want to _add_ the feature, rather than request it 

If you confirm that the two algorithms are not included in MNE-Python and agree to their inclusion, I will follow up with a proposed implementation ASAP. 

Thanks,
Matteo

larsoner, 2024-04-22:
From a very brief look it seems like SSP-SIR is a bit like `proj="reconstruct"` in https://github.com/mne-tools/mne-python/pull/8033 / [this tutorial](https://mne.tools/dev/auto_tutorials/preprocessing/50_artifact_correction_ssp.html#tut-artifact-ssp-reconstruction) but someone would have to dig into the details to see

For SOUND I'm not too familiar with how widely used it is. See also discussions of other potential algorithms like https://github.com/mne-tools/mne-python/issues/7170#issuecomment-571505186 where we suggest possible inclusion in python-meegkit or mne-sandbox to start until we see how widely used they are

matteo-d-m, 2024-05-07:
Thanks for your response. 

- `proj="reconstruct"` might indeed be some implementation of SSP-SIR, but we are trying it on some real data to check
- I believe that SOUND would be a valuable addition to MNE, but I also understand your concern. I'll just do the work and propose it to either MEEGKit or MNE-Incubator when it's done!

minsuzhang, 2024-11-18:
I'm also a TMS-EEG researcher and I double the need for @matteo-d-m 's request, thank you.