mne-tools/mne-python #12577: Inconsistent picking behavior between `Epochs` and `Evoked` in `get_data()` method

hoechenberger, 2024-04-26:
### Description of the problem

`Epochs.get_data(picks="eeg")` excludes bad channels, but `Evoked.get_data(picks="eeg")` doesn't

### Steps to reproduce

```Python
# %%
import mne

sample_dir = mne.datasets.sample.data_path()
sample_fname = sample_dir / "MEG" / "sample" / "sample_audvis_raw.fif"

raw = mne.io.read_raw_fif(sample_fname)
raw.crop(tmax=60)

events = mne.find_events(raw, stim_channel="STI 014")
raw.pick("eeg")

epochs = mne.Epochs(raw, events=events, preload=True)
evoked = epochs.average()

# %%
for inst_type, inst in zip(["epochs", "evoked"], [epochs, evoked]):
    print(f"{inst_type}:")
    print(f"    bads: {inst.info['bads']}")
    print(f"    data shape: {inst.get_data(picks='eeg').shape}")
```

### Expected results

Both methods should behave the same in regards to bad channel treatment.

### Actual results

```python
epochs:
    bads: ['EEG 053']
    data shape: (86, 59, 421)
evoked:
    bads: ['EEG 053']
    data shape: (60, 421)
```

### Additional information

This is with `main`

cbrnr, 2024-04-26:
The docstrings are identical, so I think both methods should behave identically. In addition, the docstring says:

> None (default) will pick all channels.

I think this should be:

> None (default) will pick all **good** channels.

At least that's how it currently works for `Epochs`, but not for `Evoked` apparently.

https://mne.tools/dev/generated/mne.Epochs.html#mne.Epochs.get_data
https://mne.tools/dev/generated/mne.Evoked.html#mne.Evoked.get_data

hoechenberger, 2024-04-26:
@cbrnr Note that I did **not** pass `None` in my MWE! I did pass `picks="eeg"`

larsoner, 2024-04-26:
> So something about this point is inconsistent between epochs and evoked at least when passing picks="eeg".

Neither `get_data` has an `exclude` param, sounds like one is implicitly using `exclude="bads"` and the other `exclude=()` under the hood. To change the behavior of one to match the other and be consistent I think we'd need a deprecation cycle, not sure whether or not it's worth the code churn. To update the docstring of both to reflect their actual behavior it wouldn't require a deprecation cycle. Adding `exclude` to both (to make the behavior more explicit and controllable) is probably worthwhile either way.

cbrnr, 2024-04-26:
Even if documented I think it would be a bit surprising if `Epochs` and `Evoked` used different pick defaults. So I'd document the behavior with whatever version we think makes more sense, be consistent for both methods and deprecate the inconsistent behavior. And yes, an explicit `exclude` parameter would definitely make sense to me.

hoechenberger, 2024-04-26:
I tested the MWE with Raw and now it seems it's actually Epochs who's the odd one:

```python
# %%
import mne

sample_dir = mne.datasets.sample.data_path()
sample_fname = sample_dir / "MEG" / "sample" / "sample_audvis_raw.fif"

raw = mne.io.read_raw_fif(sample_fname)
raw.crop(tmax=60)

events = mne.find_events(raw, stim_channel="STI 014")
raw.pick("eeg")

epochs = mne.Epochs(raw, events=events, preload=True)
evoked = epochs.average()

# %%
for inst_type, inst in zip(["raw", "epochs", "evoked"], [raw, epochs, evoked]):
    print(f"{inst_type}:")
    print(f"    bads: {inst.info['bads']}")
    print(f"    data shape: {inst.get_data(picks='eeg').shape}")
```
```python
raw:
    bads: ['EEG 053']
    data shape: (60, 36038)
epochs:
    bads: ['EEG 053']
    data shape: (86, 59, 421)
evoked:
    bads: ['EEG 053']
    data shape: (60, 421)
```

hoechenberger, 2024-04-26:
More digging:

- `BaseSpectrum.get_data()` features `exclude="bads'`
- `BaseTFR` ditto

hoechenberger, 2024-04-26:
My feeling is that we should add `exclude` parameters to all these `get_data()` methods if they don't have them already, retaining the current behavior; and adding a deprecation cycle for transitioning to either `exclude=()` or `exclude="bads"` for all of them

RenanFlorencio, 2025-11-06:
Hello,

Me and two friends (@luisademelo14 @fooxfeir) are taking a course and we need to choose a tool and bug to fix. We chose MNE-Python, but since we are new to the community, we decided to pick a bug marked as easy and this one seems good. 
If there's a problem with that, please let us know!

larsoner, 2025-11-25:
> Maintained Previous Behavior:

For this, rather than using `print` how about you write some tiny test on `main` that should pass that `assert`s something about shapes of returned data?

Then you add the `exclude` param defaults, and your test ensures that you haven't changed behavior.

Then you append to that same test some calls with different `exclude` values and `assert` something new/different about the obtained shapes.

This is a variant of test-driven development (strictly speaking there you'd write the whole test first even though it would fail, then modify code, but in practice it's nice to see stuff green / passing to start)

> Lastly, we would like to ask you about the following lines of code in epochs.py at the _get_data function:

I would try not to mess with this one. Internal functions get split to handle specific use cases... the `epochs` code gets used all over the place so at the end of the day having CIs/tests green should tell us if things are okay or not.

> If we were to do a deprecation cycle, we would have to think more about which one makes more sense.

Let's not do that for now... I think it would be more end-user pain that it's worth.

> Moreover, do you have any suggestions of issues we could work on?

Anything with the `EASY` label should in principle be somewhat easy, at least that's the goal 😄 

https://github.com/mne-tools/mne-python/issues?q=state%3Aopen%20label%3AEASY

closed 2026-08-28 by larsoner