mne-tools/mne-python #12581: Add option to change default channel type

PierreGtch, 2024-04-26:
### Describe the new feature or enhancement

Hi :)

At the moment, when loading an EDF raw file, the channel types are all set to EEG by default (the default type is hardcoded [here](https://github.com/mne-tools/mne-python/blob/9ba89fd030d1de30d1370631bdd3154223480be5/mne/io/edf/edf.py#L913) and [here](https://github.com/mne-tools/mne-python/blob/9ba89fd030d1de30d1370631bdd3154223480be5/mne/io/edf/edf.py#L926)).

The EDF format supports adding a prefix to the channel names to specify their types (such as `"EEG C3"`) and `mne.io.read_raw_edf()` has a parameter `infer_types` which reads this prefix, sets the type of the channel and removes the prefix from the name.

However, in many datasets, the only channels with prefixes are the EEG channels, while the channels with "rare" types do not have a prefix. In this case, even when using `infer_types=True`, all types still end up in EEG.

I propose to allow the users to specify a different default channel type when reading EDF files (or any file with unknown channel types?).

### Describe your proposed implementation

Option one, using the config:
```python
mne.set_config('DEFAULT_CHANNEL_TYPE', 'misc')
```

Easiest option to implement I think.

### Describe possible alternatives

Option two, via the function arguments:
```python
raw = mne.io.read_raw_edf(
    ...
    infer_types=True,
    default_type='misc',
)
```

### Additional context

@cbrnr Do you have an opinion on which option would be the best?

Also, are there other data formats in which the channel types can be inferred?

larsoner, 2024-04-26:
I think for a case like this you could do:
```
raw = mne.io.read_raw_edf(fname)
raw.set_channel_types({ch: "eeg" if ch.startswith("EEG") else "misc" for ch in raw.ch_names})
```
In general if such use cases can be handled by a one-liner like this it's probably better (more flexible and maintainable) to document it in the `Notes` of `read_raw_edf` for example than it would be to add more parameters to functions.

PierreGtch, 2024-04-27:
I agree this issue can be handled easily without changes to the code. But leaving things as they are makes the `infer_types` arg useless in many of its intended use cases...
Option one would require only minimal changes.

cbrnr, 2024-04-28:
Also note that `mne.io.read_raw_edf()` has `eog` and `misc` parameters, which you can use to assign these types to selected channels if you are not happy with the default EEG type.