mne-tools/mne-python #12594: `mne.io.read_raw_fil` crashes if bad channels are specified prior to import

georgeoneill, 2024-05-01:
### Description of the problem

The format used for the FIL OPM system uses a tsv file to specify channel names and statuses, so functionality to check for bad channels from the `*channels.tsv` metadata was implemented to automatically assign the bad channels on import without user intervention. Its taken me over a year to discover this part of `read_raw_fil` never worked. Sorry! 

### Steps to reproduce

```Python
import mne
import tempfile
import shutil
import os
import pandas as pd
import traceback

subject = "sub-002"
data_path = mne.datasets.ucl_opm_auditory.data_path()
opm_orig_path = (
data_path / subject / "ses-001" / "meg"
)

# Copy test data to temporary directory
temp_dir = tempfile.TemporaryDirectory()
base_dir_name = os.path.basename(opm_orig_path)
dest_dir = os.path.join(temp_dir.name, base_dir_name)
shutil.copytree(opm_orig_path, dest_dir)

opm_temp_data_path = os.path.join(dest_dir, 'sub-002_ses-001_task-aef_run-001_meg.bin')

# raw = mne.io.read_raw_fil(opm_temp_data_path)
# print('orignal attempt read in ok')

# read in channels tsv, set one to bad (G2-17-TAN) and write out
chans_tsv_path = os.path.join(dest_dir, 'sub-002_ses-001_task-aef_run-001_channels.tsv')

chans_df = pd.read_csv(chans_tsv_path, delimiter='\t')
index = chans_df[chans_df['name'] == 'G2-17-TAN'].index
chans_df.loc[index, 'status'] = 'bad'
chans_df.to_csv(chans_tsv_path, sep='\t', index=False)

try:
    raw = mne.io.read_raw_fil(opm_temp_data_path)
except Exception as e:
    traceback.print_tb(e.__traceback__)
    
temp_dir.cleanup()
```

### Expected results

An instance of `raw` should be created with the bad channels listed in the edited  `*_channels.tsv`  (this case `G2-17-TAN`) being populated in `raw.info['bads']`.

### Actual results

Instance of `raw` not created, the following traceback below

Refining sensor orientations...
[done]
  File "c:\users\george o'neill\documents\python\bad_channel_bug_test.py", line 35, in <module>
    raw = mne.io.read_raw_fil(opm_temp_data_path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<decorator-gen-383>", line 12, in read_raw_fil
  File "C:\Users\goneill\miniconda3\envs\mne\Lib\site-packages\mne\io\fil\fil.py", line 51, in read_raw_fil
    return RawFIL(binfile, precision=precision, preload=preload)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\goneill\miniconda3\envs\mne\Lib\site-packages\mne\io\fil\fil.py", line 120, in __init__
    info = _compose_meas_info(meg, chans)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\goneill\miniconda3\envs\mne\Lib\site-packages\mne\io\fil\fil.py", line 274, in _compose_meas_info
    info["bads"] = _read_bad_channels(chans)
    ~~~~^^^^^^^^
  File "C:\Users\goneill\miniconda3\envs\mne\Lib\site-packages\mne\_fiff\meas_info.py", line 1579, in __setitem__
    val = self._attributes[key](
          ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\goneill\miniconda3\envs\mne\Lib\site-packages\mne\_fiff\meas_info.py", line 992, in _check_bads
    return MNEBadsList(bads=bads, info=info)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\goneill\miniconda3\envs\mne\Lib\site-packages\mne\_fiff\meas_info.py", line 966, in __init__
    _check_bads_info_compat(bads, info)
  File "C:\Users\goneill\miniconda3\envs\mne\Lib\site-packages\mne\_fiff\meas_info.py", line 959, in _check_bads_info_compat
    raise ValueError(f"bad channel(s) {missing} marked do not exist in info")

### Additional information

It appears to be due to the fact the check for bad channels is specified before the `ch_names` has been populated. This probably can be fixed by calling `info._update_redundant()` before `_check_bads` is called.

larsoner, 2024-05-01:
Thanks for investigating, PR welcome if you're up for it @georgeoneill !

georgeoneill, 2024-05-02:
On it!

PR #12597 (2024-05-02): FIX: `mne.io.read_raw_fil` handling of bad channels 
#### Reference issue
Fixes #12594.

#### What does this implement/fix?
Changes the ordering that the `info` struct is populated in `read_raw_fil` so that bad channels can be read straight from the `channels.tsv` file without raising an error.
closed 2024-05-02 by hoechenberger