mne-tools/mne-python #12596: topoplots for OPM data

KasiaRu, 2024-05-02:
### Description of the problem

Hi there, 

The topological plotting of OPM data does not seem to work, already after selecting only one channel from a point in space.  

The is no error or crash per se, rather a figure with a black background is plotted.  

When hoovering over that figure, some channel names become visible in the bottom right, however it is not possible to magnify them as advised by the message next to the channel name nor to conduct any further action. The scale of the figure does seem to change depending on the channel selection.  

This issue is reproducible both on the MNE-available OPM data, as well as on our own UCL-acquired OPM data, and acrosss 2 machines. 

@georgeoneill has already kindly looked into it with me. 

Working topolplot is important to me as my goal is to visualise results of other analysis in this form.  

Thank you in advance for your time.

### Steps to reproduce

```Python
import matplotlib.pyplot as plt
import numpy as np

import mne

subject = "sub-002"
data_path = mne.datasets.ucl_opm_auditory.data_path()
opm_file = (
    data_path / subject / "ses-001" / "meg" / "sub-002_ses-001_task-aef_run-001_meg.bin"
)

# For now we are going to assume the device and head coordinate frames are
# identical (even though this is incorrect), so we pass verbose='error'
raw = mne.io.read_raw_fil(opm_file, verbose="error", preload=True)

# Remove all the channels with prefix RAD so only one channel exists in a point
# in space (these are dual axis sensors)
select = mne.pick_channels_regexp(raw.ch_names, regexp="^.*-RAD")
channels_to_drop = [raw.ch_names[ii] for ii in select]
raw.drop_channels(channels_to_drop)

# set flux channels to bad
bad_picks = mne.pick_channels_regexp(raw.ch_names, regexp="Flux.")
raw.info["bads"].extend([raw.ch_names[ii] for ii in bad_picks])
raw.info["bads"].extend(["G2-17-TAN"])

# HFC: include gradients by setting order to 2
projs = mne.preprocessing.compute_proj_hfc(raw.info, order=2)
raw.add_proj(projs).apply_proj(verbose="error")

# filtering
# notch
raw.notch_filter(np.arange(50, 251, 50), notch_widths=4)
# bandpass
raw.filter(2, 40, picks="meg")

# plot
events = mne.find_events(raw, min_duration=0.1)
epochs = mne.Epochs(
    raw, events, tmin=-0.1, tmax=0.4, baseline=(-0.1, 0.0), verbose="error"
)

mne.viz.plot_topo_image_epochs(epochs)
```

### Expected results

A topographic figure at least for channels one per location in space (I appreciate that how to plot channels from the same sensor is a separate issue). 

### Actual results

[image]

[image]

### Additional information

Platform             Windows-10-10.0.22631-SP0
Python               3.11.0 (main, Oct 24 2022, 18:26:48) [MSC v.1933 64 bit (AMD64)]
Executable           [c:\Users\krudzka\.venv\Scripts\python.exe](file:///C:/Users/krudzka/.venv/Scripts/python.exe)
CPU                  Intel64 Family 6 Model 151 Stepping 2, GenuineIntel (24 cores)
Memory               15.8 GB

Core
├☒ mne               1.6.0 (outdated, release 1.7.0 is available!)
├☑ numpy             1.26.4 (unknown linalg bindings (threadpoolctl module not found: No module named 'threadpoolctl'))
├☑ scipy             1.13.0
├☑ matplotlib        3.8.4 (backend=module://matplotlib_inline.backend_inline)
├☑ pooch             1.8.1
└☑ jinja2            3.1.3

Numerical (optional)
├☑ nibabel           5.2.1
└☐ unavailable       sklearn, numba, nilearn, dipy, openmeeg, cupy, pandas

Visualization (optional)
├☑ pyvista           0.43.5 (OpenGL 4.5.0 NVIDIA 472.12 via NVIDIA GeForce RTX 3080/PCIe/SSE2)
├☑ pyvistaqt         0.11.0
├☑ vtk               9.3.0
├☑ ipywidgets        8.1.2
└☐ unavailable       qtpy, ipympl, pyqtgraph, mne-qt-browser, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

drammock, 2024-10-11:
apologies for the slow reply!  Sadly I cannot replicate the problem; if I run your script on current `main` (albeit on Linux, not Windows), I see this:

[image]

Is there some chance that you have custom matplotlib `rcparams` that are affecting the underlying call to `imshow`?

larsoner, 2024-10-31:
Closing as probably fixed/working?

closed 2024-10-31 by larsoner