mne-tools/mne-python #12606: OverFlowError when calling : SourceEstimate.plot(...)

ilianAZZ, 2024-05-07:
### Description of the problem

Hello,

I'm trying to use the SourceEstimation.plot to get the 3D brain graph with the red activated areas.

When calling SourceEstimation.plot, i'm facing an Python Exception: `OverflowError: cannot convert float infinity to integer``

I'm sure the problem is from the SourceEstimate.data. Beacause after adding 1e-14 to all elements of this array, the exception is lo longer here.

My SourceEstimate.data matrix is a float64, it does not contains any INF or NAN : 

```py
print("== After editing data")
print("Shape", data_recreate.shape)
print("Min, Max", np.min(data_recreate), np.max(data_recreate))
print("Contains NaN", np.isnan(data_recreate).any())
print("Contains inf", np.isinf(data_recreate).any())
print("All types", set(type(element) for row in data_recreate for element in row))

stc_recreate = mne.SourceEstimate(data=data_recreate, vertices=vertices_recreate, tmin=tmin_recreate, tstep=tstep_recreate, subject=subject_recreate)

initial_time = 0.1
brain = stc_recreate.plot(
    subjects_dir=subjects_dir,
    initial_time=initial_time,
    smoothing_steps=7,
    hemi='both',
```

Output :
```
== After editing data
Shape (7498, 305)
Min, Max 0.0 1.8034394545323202e-10
Contains NaN False
Contains inf False
All types {<class 'numpy.float64'>}
```

I then supposed that this error was caused by the presence of "0" in the data. 
So I tried with a matrix that contains multiple lines with 0, things goes well.

```py
data_recreate = stc_copy.data.copy()
data_recreate += 1e-14   #  Avoir OverflowException

data_recreate[:, 1:10] = 0
data_recreate[0:200:, :] = 0

```

So I suppose this as an issue. Let me know if I'm doing something wrong.

Best

```
File c:\Users\Ilian\Documents\dev\best-python\.venv\lib\site-packages\mne\source_estimate.py:743, in _BaseSourceEstimate.plot(self, subject, surface, hemi, colormap, time_label, smoothing_steps, transparent, alpha, time_viewer, subjects_dir, figure, views, colorbar, clim, cortex, size, background, foreground, initial_time, time_unit, backend, spacing, title, show_traces, src, volume_options, view_layout, add_data_kwargs, brain_kwargs, verbose)
    [709](file:///C:/Users/Ilian/Documents/dev/best-python/.venv/lib/site-packages/mne/source_estimate.py:709) @copy_function_doc_to_method_doc(plot_source_estimates)
    [710](file:///C:/Users/Ilian/Documents/dev/best-python/.venv/lib/site-packages/mne/source_estimate.py:710) def plot(
    [711](file:///C:/Users/Ilian/Documents/dev/best-python/.venv/lib/site-packages/mne/source_estimate.py:711)     self,
   (...)
    [741](file:///C:/Users/Ilian/Documents/dev/best-python/.venv/lib/site-packages/mne/source_estimate.py:741)     verbose=None,
    [742](file:///C:/Users/Ilian/Documents/dev/best-python/.venv/lib/site-packages/mne/source_estimate.py:742) ):
--> [743](file:///C:/Users/Ilian/Documents/dev/best-python/.venv/lib/site-packages/mne/source_estimate.py:743)     brain = plot_source_estimates(
    [744](file:///C:/Users/Ilian/Documents/dev/best-python/.venv/lib/site-packages/mne/source_estimate.py:744)         self,
    [745](file:///C:/Users/Ilian/Documents/dev/best-python/.venv/lib/site-packages/mne/source_estimate.py:745)         subject,
    [746](file:///C:/Users/Ilian/Documents/dev/best-python/.venv/lib/site-packages/mne/source_estimate.py:746)         surface=surface,
    [747](file:///C:/Users/Ilian/Documents/dev/best-python/.venv/lib/site-packages/mne/source_estimate.py:747)         hemi=hemi,
    [748](file:///C:/Users/Ilian/Documents/dev/best-python/.venv/lib/site-packages/mne/source_estimate.py:748)         colormap=colormap,
[... 207 lines cut ...]
CPU                  Intel64 Family 6 Model 165 Stepping 2, GenuineIntel (8 cores)
Memory               31.8 GB

Core
├☑ mne               1.7.0 (latest release)
├☑ numpy             1.26.4 (OpenBLAS 0.3.23.dev with 8 threads)
├☑ scipy             1.13.0
└☑ matplotlib        3.8.4 (backend=module://matplotlib_inline.backend_inline)

Numerical (optional)
├☑ sklearn           1.4.2
├☑ nibabel           5.2.1
├☑ nilearn           0.10.4
├☑ pandas            2.2.2
└☐ unavailable       numba, dipy, openmeeg, cupy, h5io, h5py

Visualization (optional)
├☑ pyvista           0.43.5 (OpenGL 4.5.0 NVIDIA 552.12 via NVIDIA GeForce GTX 1650 Ti/PCIe/SSE2)
├☑ pyvistaqt         0.11.0
├☑ vtk               9.3.0
├☑ qtpy              2.4.1 (PyQt6=6.6.1)
└☐ unavailable       ipympl, pyqtgraph, mne-qt-browser, ipywidgets, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo, eeglabio, edfio, mffpy, pybv

drammock, 2024-05-14:
Thanks for the report. It appears it's happening because the maximum value of your data is zero (which may indicate a mistake somewhere in your script; is data with no positive values plausible for you?)

The error is happening here:

https://github.com/mne-tools/mne-python/blob/1c5b39ff1d99bbcb2fc0e0071a989b3f3845ff30/mne/viz/_brain/_brain.py#L4153-L4171

@wmvanvliet you wrote this code in your UI event system work (#11891), any chance you have time to take a look?

wmvanvliet, 2024-05-15:
Ah, the np.log10() call needs protection to prevent no.log10(0) of course. My bad :) PR incoming

PR #12612 (2024-05-16): Fix Brain slider ranges
This updates the logic that determines the ranges for the three sliders (fmin, fmid, fmax) on the left of a `Brain` figure. There was an overflow when the data is all zeros (`np.log10(fmax)`) which this PR fixes by using `np.log10(max(fmax, 1e-30))` instead. While I was there, I noticed that the minimum and maximum values for the sliders were set based on the `fmin` and `fmax` of the current timepoint instead of the entire data. Finally, on second thought, the minimum value for the sliders should always be `0`, that makes a lot more sense to me than capping it to `fmin` [... cut]
closed 2024-05-16 by wmvanvliet