mne-tools/mne-python #12608: Add option to import BrainVision files without type in markers

cbrnr, 2024-05-10:
BrainVision markers (stored in `.vmrk` files) consist of the following fields:

```
Mk<number>=<Type>,<Description>,<Position>,<Size>,<Channel>,<Date>
```

For example:

```
Mk2=Comment,bad_segment,251,391,0
```

This is a marker as exported via `raw.export()` using `pybv` from an annotation with the description `bad_segment`. Since the `Type` field must be present, `pybv` uses `Comment` by default.

However, when importing the exported BrainVision file, annotations are created from markers, and the description will be pieced together as `<Type>/<Description>`, i.e. `Comment/bad_segment` in this case.

This makes the round-trip inconsistent, because I'd like to recover the original annotation descriptions from markers. This simply means that I'd need an option for `read_raw_brainvision()` to ignore the `Type` fields when constructing annotations. This would require a new parameter, e.g. `ignore_marker_types=False` (or something along these lines).

Would that be an addition that everyone is OK with (pinging @sappelhoff explicitly as one of the `pybv` maintainers)?

hoechenberger, 2024-05-12:
> ignore_marker_types=False

It could make sense to support a list of strings here too, wdyt?

cbrnr, 2024-05-12:
> It could make sense to support a list of strings here too, wdyt?

You mean to ignore only specific types, e.g. `ignore_marker_types=["Comment", "Stimulus"]`? I'm not sure how often this is needed, but of course it would be no problem to implement it. I was thinking of just one use case where ignoring types could be needed, namely importing a file that was exported by MNE, in which case all markers will have the `Comment` type. Since this is a bit arbitrary, it makes sense to remove those "artificial" types on import. In contrast, "native" BrainVision files (e.g. those created by BrainVision Analyzer) will contain various marker types and descriptions in general, so I'd want to keep those (at least by default).

drammock, 2024-05-13:
> You mean to ignore only specific types, e.g. `ignore_marker_types=["Comment", "Stimulus"]`? I'm not sure how often this is needed

Seems to me like there's clear motivation for "ignore marker types when they're all identical" and uncertain motivation for "let me specify which one(s) to ignore". I'd suggest to assume YAGNI on the second one, and we can overload the param later (to accept list-likes) if folks ask for it.

I'd also consider something like an "always/auto/never" pattern, where "auto" means "ignore them IFF they're all identical". But a simple boolean seems adequate for now --- again, we can update the param later to give more flexibility if folks ask for it.

sappelhoff, 2024-05-13:
sounds good to me, and agreed with YAGNI for the second suggestion.

hoechenberger, 2024-05-13:
agreed 👍

PR #12652 (2024-06-12): Add option to ignore BrainVision marker types
Fixes #12608.
closed 2024-06-12 by larsoner