mne-tools/mne-python #12611: BUG: Incompatibility of `sample_weight` with sklearn 1.4+

larsoner, 2024-05-15:
See TODOs in https://github.com/mne-tools/mne-python/pull/12610/files and the [linked sklearn docs](https://scikit-learn.org/stable/metadata_routing.html). TL;DR is that `logreg = OneVsRestClassifier(LogisticRegression(...))` is preferred now over `logreg = LogisticRegression(..., multi_class="ovr")` but then doing `logreg.fit(X, Y, sample_weight=...)` is an error because you have to enable metadata routing. I tried doing this with a context manager to set and restore the state but couldn't get it to work easily, especially with our sliding estimators.

Maybe one for @agramfort :slightly_smiling_face:

agramfort, 2024-05-15:
do you have a small snippet to replicate the pb or a specific pytest command?

larsoner, 2024-05-15:
Remove this conditional you will see the error

https://github.com/mne-tools/mne-python/blob/e43b5d897228df7de41799583df2e38adbcffc6e/mne/decoding/tests/test_search_light.py#L213

PR #12620 (2024-05-20): fix for sklearn metadata
fixes #12611
closed 2024-05-20 by larsoner