mne-tools/mne-python #12621: Eyelink: Accept blank recording dates?

sappelhoff, 2024-05-21:
I have eyelink data (`.asc`) where I "deidentified" the header by blanking out some data, see this example:

```
** CONVERTED FROM edf using edfapi 4.2.1.0 Windows standalone Jun 19 2021
** DATE:
** TYPE: EDF_FILE BINARY EVENT SAMPLE TAGGED
** VERSION: EYELINK II 1
** SOURCE: EYELINK CL
** EYELINK II CL v5.15 Jan 24 2018
** CAMERA: Eyelink GL Version 1.2 Sensor=AG7
** SERIAL NUMBER:
** CAMERA_CONFIG:
** RECORDED BY ...
**
```

1. "CONVERTED FROM" was adjusted to no longer name the original data file from which the conversion was performed (as it included a recording date)
2. "DATE" was blanked out, as were "SERIAL NUMBER" and "CAMERA_CONFIG"

I have performed "blanking out" in discussion with SR-Research (who actually also supplied a `sed` based script that I could use on EDF files directly), and I assume therefore that their tools won't have problems with this "blanking out".

However, MNE-Python does throw an error:

```
[331](file:///C:/Users/stefan/miniconda3/envs/pmetrics_analysis/Lib/_strptime.py:331) found = format_regex.match(data_string)
    [332](file:///C:/Users/stefan/miniconda3/envs/pmetrics_analysis/Lib/_strptime.py:332) if not found:
--> [333](file:///C:/Users/stefan/miniconda3/envs/pmetrics_analysis/Lib/_strptime.py:333)     raise ValueError("time data %r does not match format %r" %
    [334](file:///C:/Users/stefan/miniconda3/envs/pmetrics_analysis/Lib/_strptime.py:334)                      (data_string, format))
    [335](file:///C:/Users/stefan/miniconda3/envs/pmetrics_analysis/Lib/_strptime.py:335) if len(data_string) != found.end():
    [336](file:///C:/Users/stefan/miniconda3/envs/pmetrics_analysis/Lib/_strptime.py:336)     raise ValueError("unconverted data remains: %s" %
    [337](file:///C:/Users/stefan/miniconda3/envs/pmetrics_analysis/Lib/_strptime.py:337)                       data_string[found.end():])

ValueError: time data '' does not match format '%a %b %d %H:%M:%S %Y'
```

If I edit my ASC file header to include a `"DATE"` again (of this form: `Mon Jan 01 12:12:12 2024`), then everything works.

I think we should be able to deal with blanked out dates, and assign a missing measurement date.

Thoughts? @scott-huberty

larsoner, 2024-05-21:
Agreed if you use the manufacturer suggested anonymization method it should work. Here I think if the date string is empty the meas date should be set to None

scott-huberty, 2024-05-21:
Agreed, as this has come up before https://github.com/mne-tools/mne-python/issues/12516#issuecomment-2023174266 . I can address it in #12516 unless you beat me to it @sappelhoff ! (I probably won't be able to get to it for a couple weeks).

sappelhoff, 2024-05-21:
Great, thanks @scott-huberty! :-)

PR #12687 (2024-06-28): FIX: Gracefully handle missing datetime in Eyelink File
👋 

Fixes #12621
closed 2024-06-28 by larsoner