mne-tools/mne-python #12630: Add an SSVEP classification algorithm as a new feature

SparklingWasser, 2024-05-28:
### Describe the new feature or enhancement

Hi,

I recently have been looking for a Python library for SSVEP classification, but it seems that there is no such major one offering this feature. Thus, I am writing a function myself and was wondering if it would be okay to include this to MNE, for people like me to benefit from this. 

Currently, the most of the public scripts for SSVEP classification are written in MATLAB, and the only SSVEP classification method in MNE (using `epochs.compute_psd`) is based on PSD, which is suitable to grasp a basic concept of SSVEP classification but not practical.

### Describe your proposed implementation

Canonical correlation analysis (CCA) [(Lin 2006)](https://doi.org/10.1109/TBME.2006.886577) and multivariate synchronization index (MSI) [(Zhang 2014)](https://doi.org/10.1016/j.jneumeth.2013.07.018) are the two decent methods to classify SSVEP responses without requiring training dataset. These find a spatial filter to maximize the correlation between multi-channel EEG data and the reference signals (sine or cosine waves with stimulation frequencies). CCA is probably the most popular algorithm for SSVEP classification, and also most of the state-of-the-art algorithms are based on this [(Wong 2021](https://doi.org/10.1109/TBME.2021.3133594), [Nakanishi 2018)](https://doi.org/10.1109/TBME.2017.2694818). MSI is less popular, but it always showed a better performance in my experience. 

I guess this would best fit under `mne.decoding` as a new class. The name can be `mne.decoding.SSVEPClassification`, for example. It would process `mne.Epochs` and yield the stimulation index as an output.

### Describe possible alternatives

To the best of my knowledge, it is best option to create a new class to add this feature.

agramfort, 2024-05-28:
do you have a public dataset we could use to build the example? I
would start by adding the dataset fetcher for this so it becomes easier to
evaluate on the same data and also on our CIs.

> Message ID: ***@***.***>
>

SparklingWasser, 2024-05-29:
Can we use a public [benchmark dataset](http://bci.med.tsinghua.edu.cn/download.html) by [Wang 2016](https://doi.org/10.1109/TBME.2017.2694818)? It is widely used one in the field, but is written in .mat format and is already epoched (no pure raw dataset).
Data from one participant (around 100 MB) is enough.