mne-tools/mne-python #12639: Tracking issue for (implicit) MNE dependencies still lacking binary wheels for `aarch64` (ARM Linux)

hoechenberger, 2024-06-01:
This is just to help me (and you? 😄) track which packages are still lacking binary wheels for `aarch64`, hence preventing us from providing an installation that solely relies on PyPI / `pip`:

- [ ] `h5py`: https://github.com/h5py/h5py/pulls
- [ ] `cftime`: https://github.com/Unidata/cftime/issues/333
- [ ] `nitime`: https://github.com/nipy/nitime/issues/216
- [ ] `openmeeg`: https://github.com/openmeeg/openmeeg/pull/664, https://github.com/openmeeg/openmeeg/pull/666 (waiting for new release)
- [ ] `psutil`: https://github.com/giampaolo/psutil/issues/2206
- [ ] `vtk`: https://gitlab.kitware.com/vtk/vtk/-/issues/18648
- [ ] `pyqt`: https://www.riverbankcomputing.com/pipermail/pyqt/2022-October/044933.html

hoechenberger, 2024-06-03:
@larsoner I see you release `openmeeg` 2.5.11 last week, but this hasn't been published to PyPI yet, is that intentional?

larsoner, 2024-06-03:
Nice catch, there was a spurious failure in the tag build, I restarted and now [2.5.11 is on PyPI](https://pypi.org/project/openmeeg/2.5.11/#files)!

hoechenberger, 2024-06-03:
@larsoner Great, thanks! 😊

hoechenberger, 2024-10-07:
The only remaining one is VTK, and it doesn't look like we will see `aarch64` wheels for it anytime soon.